SysSec-KAIST/LTESniffer #11: Utilize 2x b210

grmono, 2023-04-28:
Could we utilize 2x b120 or b205mini connected with a ref clock to run uplink and downlink?

What changes are necessary to accomplish this and if we define a second rf device in LTESniffer_core and configure it on the appropriate rx freq etc will that be enough or will further changes be required deeper in the codebase?

mohanex, 2023-05-01:
This project just got released, it is somehow based of Falcon project by falkenber9.
I didn't test it with the b205mini but Falcon is working fine with it so I might say that yes you could utilize the b205mini for the uplink and downlink.
If you tried it, and didnt go well correct me then.

hdtuanss, 2023-05-01:
LTESniffer currently does not support utilizing 2x USRP B210 or B205 mini for uplink sniffing. This is because its synchronization mechanism is based on "srsran_rf_set_rx_freq" function, which only can control 1 USRP in the current release. If you want to use 2x USRP B210 simultaneously, you might need to check whether "srsran_rf_set_rx_freq" function supports controlling 2x USRP at the same time or not. Then, you might need to modify LTESniffer_core.cc to control and synchronize with 2x USRPs. Note that even if you use the same ref clock for 2x USRPs, you might want to have an additional function to synchronize uplink and downlink subframe timing.

closed 2023-05-01 by hdtuanss