SysSec-KAIST/LTESniffer #17: Missing paging requests

grmono, 2023-05-05:
I am using the downlink sniffer with a b205mini and compared to the basic pdsch_ue example already present in srsran it seems that the implementation is missing quote alot of paging requests.

The pdsch_ue is run with rnti 0xffff, and when comparing the outputed pcaps, LTESniffer only shows a few paging requests meanwhile pdsch_ue has multiple.

Are the missing paging requests related to decoding failures or some other issue with rntis being looped or something else?

hdtuanss, 2023-05-06:
Hi grmono, could you show the your screen log when running LTESniffer?
I'm not sure whether pdsch_ue.cc supports saving decoded packets to pcap file or not.
In my environment, I have similar paging results from LTESniffer and AirScope.
If you can, please run LTESniffer with B210 and check the result again.
Thanks

grmono, 2023-05-07:
I currently do not have a b210 to test it but my understanding and correct me if i am wrong is that the downlink sniffer requires only 1 rx antenna.
For pdsch_ue.cc you can inject packets to the network with -u parameter and i have made my own decoder using pycrate to decode the paging requests.

I compared both implementations and LTESniffer is definitely missing many paging requests atleast when used with b205/b200 with single rx antenna, you can use pdsch_ue with -u 5555 -r 0xffff option this will inject the paging packets in your localhost at port 5555 which you can capture with wireshark for your test. 

Wireshark wont be able to decode them since they are missing the proper mac header but every packet you will see is essentially a paging request if you want to compare (this is also how old LTE-Sniffer was build available publicly)

If rx 2 antennas are necessary then what is the reason exactly?

hdtuanss, 2023-05-08:
Hi, thanks for your information.
Decoding DL paging messages do not need 2 RX antennas. The reason I mentioned using USRP B210 with 2 antennas is to examine whether the missing paging is because of USRP or not, as I only tested on USRP B210. 

I borrowed a USRP B200 and tested decoding paging messages with B200. I confirm that using B200 with 1 antenna returns some missing paging messages when compared with AirScope. The problem might come from the algorithm of LTESniffer when using 1 antenna for downlink sniffing. However, when using USRP B210 with 2 RX antenna, LTESniffer has the same number of paging messages as AirScope. I will fix this problem soon. At the same time, it would be better if you can use USRP B210 or other USRP with 2 RX antennas to test.

grmono, 2023-05-08:
Great could you point me out to where u think the problem is exactly in the code i could assist in fixing the problem or performing more tests

hdtuanss, 2023-05-08:
I fixed the error, please look at line 755 in DL_Sniffer_PDSCH.cc, I added a filter to accept all DCI with paging RNTI (65534). Please pull repo and test again.
Thanks a lot.

grmono, 2023-05-08:
I retested again and it seems to be capturing a lot more paging requests, i will compare with pdsch_ue and report back the comparison, thank you

closed 2023-05-12 by hdtuanss