SysSec-KAIST/LTESniffer #21: I/Q data for tests that contain DL and UL

Novitoll, 2023-05-18:
Hello,

Could you please provide with the raw IQ data (in eCPRI PCAP or CSV or raw binary as some complex doubles/floats format) for tests?
This is important for the case when you don't have the 2nd SDR to capture UL in case of FDD.

I have only 1 SDR and can capture DL frequency, 2 RX antennas, but unfortunately, there is no TDD tower around, so another SDR is needed to capture UL frequency. And plus synchronization is an issue in case 2 SDRs are from 2 different vendors.

So if you can provide an example of IQ data (TM1, TM2, TM3-4) that contain in-sync (or it's can be sync-ed via PSS/SSS time offset, right?) -- that would be a really really helpful.

Thanks!

hdtuanss, 2023-05-21:
Hi, I cannot provide I/Q sample file because it contains user data in my country. However, I will try to record several I/Q sample files from srseNB and upload them as soon as I can.

closed 2023-05-21 by hdtuanss

Novitoll, 2023-05-21:
@hdtuanss , Hello, ok, please ping me either here or via snovitoll@gmail.com, please make sure that both DL & UL data are there. Thanks! Ideally, with MIMO and with TMs that require UE feedback (TM3-6)

sabr-fmad, 2023-05-29:
@hdtuanss , hello, any updates?

Novitoll, 2023-05-29:
@hdtuanss , any updates?

hdtuanss, 2023-05-30:
@Novitoll Hi, sorry for being late. I'm pretty busy this time with a conference, I will make it for you next week