SysSec-KAIST/LTESniffer #22: Issue

flyguy999999, 2023-05-21:
Hello when I started using this program yesterday I was recieving several paging request, now I am recieving none and just get the following:

-------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[09:35:52] Processed 1000/1000 subframes
[09:35:53] Processed 1000/1000 subframes
[09:35:54] Processed 1000/1000 subframes
[09:35:55] Processed 1000/1000 subframes
[09:35:56] Processed 1000/1000 subframes
[09:35:57] Processed 1000/1000 subframes
[09:35:58] Processed 1000/1000 subframes
[09:35:59] Processed 1000/1000 subframes
[09:36:00] Processed 1000/1000 subframes
[09:36:01] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[09:36:02] Processed 1000/1000 subframes
[09:36:03] Processed 1000/1000 subframes
[09:36:04] Processed 1000/1000 subframes
[09:36:05] Processed 1000/1000 subframes
[09:36:06] Processed 1000/1000 subframes
[09:36:07] Processed 1000/1000 subframes
[09:36:08] Processed 1000/1000 subframes
[09:36:09] Processed 1000/1000 subframes
[09:36:10] Processed 1000/1000 subframes
[09:36:11] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[09:36:12] Processed 1000/1000 subframes
[09:36:13] Processed 1000/1000 subframes
[09:36:14] Processed 1000/1000 subframes
[09:36:15] Processed 1000/1000 subframes
[09:36:16] Processed 1000/1000 subframes
[09:36:17] Processed 1000/1000 subframes

Do you know what the issue could be?

hdtuanss, 2023-05-21:
Hi, could you provide more information about SDR, CPU, and the cell you are listening? It would be great if you can provide the whole screen log. One possible reason is no user connected to the cell you are listening

flyguy999999, 2023-05-21:
I am using a b200 with the command "sudo ./LTESniffer -A 1 -W 4 -f 2140e6 -C -m 0" This DL pulls several paging when I use DLProbe or "pdsch_ueA" so users are connecting 
the log is:

[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning receiver to 2150.000 MHz
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 410 FDD, CP: Normal  , DetectRatio=100% PSR=3.76, Power=-52.9 dBm
Decoding PBCH for cell XXX (N_id_2=2)  "editted the cell"
Setting sampling rate 11.52 MHz
Finding PSS... Peak: 1.10425, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.04319, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.07782, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0176, FrameCnt: 0 State: 0
Finding PSS... Peak: 2.23485, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.17778, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.03403, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.02215, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.57036, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.07246, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.06914, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00928, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.2256, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.19314, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.6503, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.12424, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0586, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.23168, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00547, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.54834, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.44024, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.10073, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.04928, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.07058, FrameCnt: 0 State: 0
Finding PSS... Peak: 3.10622, FrameCnt: 0 State: 1
 - Type:            FDD
 - PCI:             XXX
 - Nof ports:       4
 - CP:              Normal  
 - PRB:             50
 - PHICH Length:    Normal
 - PHICH Resources: 1
 - SFN:             700
Decoded MIB. SFN: 700, offset: 2
[11:35:30] Processed 1000/1000 subframes
[11:35:31] Processed 1000/1000 subframes
[11:35:32] Processed 1000/1000 subframes
[11:35:33] Processed 1000/1000 subframes
[11:35:34] Processed 1000/1000 subframes
[11:35:35] Processed 1000/1000 subframes
[11:35:36] Processed 1000/1000 subframes
[11:35:37] Processed 1000/1000 subframes
[11:35:38] Processed 1000/1000 subframes
[11:35:39] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[11:35:40] Processed 1000/1000 subframes
[11:35:41] Processed 1000/1000 subframes
[11:35:42] Processed 1000/1000 subframes
[11:35:43] Processed 1000/1000 subframes
[11:35:44] Processed 1000/1000 subframes
[... 37 lines cut ...]
[11:36:06] Processed 1000/1000 subframes
[11:36:07] Processed 1000/1000 subframes
[11:36:08] Processed 1000/1000 subframes
[11:36:09] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[11:36:10] Processed 1000/1000 subframes
[11:36:11] Processed 1000/1000 subframes
[11:36:12] Processed 1000/1000 subframes
[11:36:13] Processed 1000/1000 subframes
[11:36:14] Processed 1000/1000 subframes
[11:36:15] Processed 1000/1000 subframes
[11:36:16] Processed 1000/1000 subframes
[11:36:17] Processed 1000/1000 subframes
[11:36:18] Processed 1000/1000 subframes
[11:36:19] Processed 1000/1000 subframes
------------------------------------------------------------------------------------------------------

with nothing going to the .pcap file.

hdtuanss, 2023-05-21:
You are listening on a 4-port cell (nof ports: 4) which is not supported by LTESniffer, please try again on a cell with nof ports less than 2.