SysSec-KAIST/LTESniffer #26: Read IQ from CSV or binary file instead of SDR

Novitoll, 2023-06-03:
I dont see that the code is able to receive IQ from CSV or binary (raw floats or doubles) file.
Could you please provide the sample code or the code that needs to be changed to read IQ stream from the input file?

Novitoll, 2023-06-04:
I have following CSV (captured from bladeRF):

```
-0.2041015625,-0.0791015625
-0.05224609375,-0.0751953125
0.06640625,-0.1298828125
0.09326171875,0.02294921875
0.1474609375,-0.05859375
...
```

and I am using this Python script to convert each IQ to float bytes, as in srsRAN it's read as

`if (srsran_filesource_init(&q->file_source, file_name, SRSRAN_COMPLEX_FLOAT_BIN)) {`

```python
import sys
import struct

f_csv = sys.argv[1]

with open(f_csv, "r") as f:
    lines = f.readlines()

with open("data.bin", "wb+") as f_w:
    for line in lines:
        i, q = line.split(',')

        i = struct.pack("<f", float(i))
        q = struct.pack("<f", float(q))

        f_w.write(q)
        f_w.write(i)
```

and no data is decoded (I've checked with `-v`, and MIB is not decoded neither). @hdtuanss , Could you please suggest?

```
$ ./src/LTESniffer -A 1 -f 1870M -m 0 -P 1 -c 151 -i ~/Downloads/data.bin

LTESniffer

Sun_Jun__4_15.07.35_2023.
Creating Phy
Creating 4 Worker threads 
[15:07:36] Processed 1000/1000 subframes
[15:07:36] Processed 1000/1000 subframes
[15:07:37] Processed 1000/1000 subframes
[15:07:37] Processed 1000/1000 subframes
[15:07:37] Processed 1000/1000 subframes
[15:07:38] Processed 1000/1000 subframes
[15:07:38] Processed 1000/1000 subframes
[15:07:38] Processed 1000/1000 subframes
[15:07:39] Processed 1000/1000 subframes
[15:07:39] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

[15:07:40] Processed 1000/1000 subframes
[15:07:40] Processed 1000/1000 subframes
[15:07:40] Processed 1000/1000 subframes
[15:07:41] Processed 1000/1000 subframes
[15:07:41] Processed 1000/1000 subframes
[15:07:41] Processed 1000/1000 subframes
[15:07:42] Processed 1000/1000 subframes
[15:07:42] Processed 1000/1000 subframes
[15:07:42] Processed 1000/1000 subframes
[15:07:43] Processed 1000/1000 subframes
```

hdtuanss, 2023-06-04:
Hi, option -I is for inputting an IQ file that contains aligned subframes and starts with a subframe with index 0 (containing MIB). The I/Q file can be generated using the FalconCaptureProbe tool from FALCON (https://github.com/falkenber9/falcon). Just so you know, this tool only supports capturing subframes using 1 antenna. In your case, your I/Q sample file might not contain aligned subframes with a subframe index 0 at the start, so LTESniffer cannot decode your I/Q file successfully. Also, this option is used for only testing, and it does not support full functionalities as direct decoding from the SDR of LTESniffer. I recommend you modify the code of FalconCaptureProbe tool from FALCON for capturing with 2 RX antennas and then decode by LTESniffer.
Thanks

closed 2023-06-06 by Novitoll