SysSec-KAIST/LTESniffer #28: LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()

CumtZhangkuiyuan, 2023-06-24:
Hello, when I was executing the downlink signal sniffing command, the following error occurred:

/LTESniffer/buiid/srsRAN src/lib/src/phy/rf/rf_ Uhd_ Imp. cc: 1340: Erroneous out while receiving samples from UHD

/LTESniffer/build/srsRAN src/lib/src/phy/ue/ue_ Sync. c: 772: Error receiving samples

/LTESniffer/src/src/LTESniffer Core. cc: 370: Error calling srsrsran_ Ue_ Sync_ Work()

Our environment configuration completely meets the requirements, but it still cannot run. Can you tell me where the problem is?

hdtuanss, 2023-06-24:
Hi, could you include the whole terminal log and your environment, and hardware setup?

CumtZhangkuiyuan, 2023-06-25:
Hi, I am very pleased to receive your reply. Our hardware uses USRP X310 with 2 daughterboards and 2 omnidirectional antennas. 
The whole terminal log is as follow:

sudo ./LTESniffer -A 2 -W 4 -f 1850e6 -u 1755e6 -C -m 1

LTESniffer

Sun_Jun_25_11.00.24_2023.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD  zmq 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.heads-v4.0.0.0-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning DL receiver to 1850.000 MHz
Tunning UL receiver to 1755.000 MHz
Searching for cell...
 Found Cell_id: 435 FDD, CP: Normal  , DetectRatio=100% PSR=2.61, Power=3.3 dBm
*Found Cell_id: 124 FDD, CP: Normal  , DetectRatio=90% PSR=2.77, Power=5.2 dBm
 Found Cell_id:   5 FDD, CP: Normal  , DetectRatio=67% PSR=2.34, Power=-1.9 dBm
Decoding PBCH for cell 124 (N_id_2=1)
Setting sampling rate 23.04 MHz
Finding PSS... Peak: 2.78051, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0367, FrameCnt: 0 State: 0
Finding PSS... Peak: 5.79223, FrameCnt: 0 State: 1
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.

However, when LTESniffer does not do the cell search but connects directly to the specified cell, it can run normally, but no data can be collected. The whole terminal log is as follow:

sudo ./LTESniffer -A 2 -W 4 -f 1850e6 -u 1755e6 -I 435 -m 1

LTESniffer

Sun_Jun_25_11.04.46_2023.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD  zmq 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.heads-v4.0.0.0-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[... 67 lines cut ...]
[11:05:04] Processed 1000/1000 subframes
[11:05:05] Processed 1000/1000 subframes
[11:05:06] Processed 1000/1000 subframes
[11:05:07] Processed 1000/1000 subframes
[11:05:08] Processed 1000/1000 subframes
[11:05:09] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success  SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success  SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are max 16QAM, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown 

^CLTESniffer_Core: Exiting...
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are max 16QAM table, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown 

Destroyed Phy
Skipped subframe: 0 / 10709

CumtZhangkuiyuan, 2023-06-25:
Hi, I am very pleased to receive your reply. Our hardware uses USRP X310 with 2 daughterboards and 2 omnidirectional antennas.
The whole terminal log is as follow:

sudo ./LTESniffer -A 2 -W 4 -f 1850e6 -u 1755e6 -C -m 1

LTESniffer

Sun_Jun_25_11.00.24_2023.
Creating Phy
Creating 4 Worker threads
Opening RF device with 2 RX antennas...
Available RF device list: UHD zmq
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.heads-v4.0.0.0-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP] 1) catch time transition at pps edge
[INFO] [MULTI_USRP] 2) set times next pps (synchronously)
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning DL receiver to 1850.000 MHz
Tunning UL receiver to 1755.000 MHz
Searching for cell...
Found Cell_id: 435 FDD, CP: Normal , DetectRatio=100% PSR=2.61, Power=3.3 dBm
*Found Cell_id: 124 FDD, CP: Normal , DetectRatio=90% PSR=2.77, Power=5.2 dBm
Found Cell_id: 5 FDD, CP: Normal , DetectRatio=67% PSR=2.34, Power=-1.9 dBm
Decoding PBCH for cell 124 (N_id_2=1)
Setting sampling rate 23.04 MHz
Finding PSS... Peak: 2.78051, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0367, FrameCnt: 0 State: 0
Finding PSS... Peak: 5.79223, FrameCnt: 0 State: 1
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.

However, when LTESniffer does not do the cell search but connects directly to the specified cell, it can run normally, but no data can be collected. The whole terminal log is as follow:

sudo ./LTESniffer -A 2 -W 4 -f 1850e6 -u 1755e6 -I 435 -m 1

LTESniffer

Sun_Jun_25_11.04.46_2023.
Creating Phy
Creating 4 Worker threads
Opening RF device with 2 RX antennas...
Available RF device list: UHD zmq
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.heads-v4.0.0.0-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP] 1) catch time transition at pps edge
[INFO] [MULTI_USRP] 2) set times next pps (synchronously)
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[... 59 lines cut ...]
Finding PSS... Peak: 1.01894, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.27664, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.59707, FrameCnt: 0 State: 0
Finding PSS... Peak: 4.35041, FrameCnt: 0 State: 1
[11:05:00] Processed 1000/1000 subframes
[11:05:01] Processed 1000/1000 subframes
[11:05:02] Processed 1000/1000 subframes
[11:05:03] Processed 1000/1000 subframes
[11:05:04] Processed 1000/1000 subframes
[11:05:05] Processed 1000/1000 subframes
[11:05:06] Processed 1000/1000 subframes
[11:05:07] Processed 1000/1000 subframes
[11:05:08] Processed 1000/1000 subframes
[11:05:09] Processed 1000/1000 subframes
Num RNTI Max Mod Active Success SNR(dB) DL-UL_delay(us) Other_Info
Num RNTI Max Mod Active Success SNR(dB) DL-UL_delay(us) Other_Info
[256Tracking] Total: 0 RNTIs are max 16QAM, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown

^CLTESniffer_Core: Exiting...
Num RNTI Max Mod Active Success SNR(dB) DL-UL_delay(us) Other_Info
Num RNTI Max Mod Active Success SNR(dB) DL-UL_delay(us) Other_Info
[256Tracking] Total: 0 RNTIs are max 16QAM table, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown

Destroyed Phy
Skipped subframe: 0 / 10709

hdtuanss, 2023-06-25:
Thanks for your information. Could you check the number of Physical Resource Block (PRB) of the listening cell and try this command again: sudo ./LTESniffer -A 2 -W 4 -f 1850e6 -u 1755e6 -I 435 -p <PRB> -m 1.
The specified cell function needs the PRB to run.
Also, it would be great if you can try on another cell.
I haven't seen this problem before.

CumtZhangkuiyuan, 2023-06-29:
Yes,  I try this command again, but I'm not seeing any packets being captured and ltesniffer_ul_mode.pcap is empty. The whole terminal log is as follow:

sudo ./LTESniffer -A 2 -W 4 -f 1815.1e6 -u 1720.1e6 -I 61 -p 10 -m 1

LTESniffer

Thu_Jun_29_19.17.03_2023.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.1.0.heads-v4.1.0.0-0-gd21735d5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [GPS] No GPSDO found
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning DL receiver to 1815.100 MHz
Tunning UL receiver to 1720.100 MHz
Setting sampling rate 3.84 MHz
Finding PSS... Peak: 1.02761, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.15361, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.49319, FrameCnt: 0 State: 0
Finding PSS... Peak: 6.60287, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.06578, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.02492, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.52723, FrameCnt: 0 State: 0
Finding PSS... Peak: 9.14575, FrameCnt: 0 State: 1
 - Type:            FDD
 - PCI:             61
 - Nof ports:       2
 - CP:              Normal  
 - PRB:             100
 - PHICH Length:    Normal
 - PHICH Resources: 1
 - SFN:             976
Decoded MIB. SFN: 976, offset: 0
[19:17:14] Processed 1000/1000 subframes
[19:17:15] Processed 1000/1000 subframes
[19:17:16] Processed 1000/1000 subframes
[19:17:17] Processed 1000/1000 subframes
[19:17:18] Processed 1000/1000 subframes
[19:17:19] Processed 1000/1000 subframes
[19:17:20] Processed 1000/1000 subframes
[19:17:21] Processed 1000/1000 subframes
[19:17:22] Processed 1000/1000 subframes
[19:17:23] Processed 1000/1000 subframes
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success  SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success  SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are max 16QAM, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown 

[19:17:24] Processed 1000/1000 subframes
[19:17:25] Processed 1000/1000 subframes
^CLTESniffer_Core: Exiting...
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are max 16QAM table, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown 

Destroyed Phy
Skipped subframe: 0 / 12200
nof_decoded_locations, nof_cce, nof_missed_cce, nof_subframes, nof_subframe_collisions_dw, nof_subframe_collisions_up, time, nof_locations
19929, 88900, 1401, 12125, 0, 0, 0.000000, 159907
Skipped subframes: 0 (0%)
Saving MAC PCAP file
Deleted DL Sniffer core

hdtuanss, 2023-06-29:
Could you execute the following commands before you run LTESniffer?
```
sudo sysctl -w net.core.rmem_max=33554432
sudo sysctl -w net.core.wmem_max=33554432
sudo ifconfig <10Gb card interface> mtu 9000
sudo uhd_usrp_probe
```
After that, please modify your command to:
```
sudo ./LTESniffer -A 2 -W 4 -f 18151e5 -u 17201e5 -I 61 -p 100 -m 1
```
The number of PRB should be 100, not 10.
Thanks.

CumtZhangkuiyuan, 2023-06-30:
OK, thank you very much, I have executed the above commands before run LTESniffer. And I have modified the command to

sudo ./LTESniffer -A 2 -W 4 -f 1815.1e6 -u 1720.1e6 -I 61 -p 100 -m 1

The previous error occurred again, the terminal information are as follow:

LTESniffer

Fri_Jun_30_08.55.41_2023.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.1.0.heads-v4.1.0.0-0-gd21735d5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning DL receiver to 1815.100 MHz
Tunning UL receiver to 1720.100 MHz
Setting sampling rate 23.04 MHz
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
Finding PSS... Peak: 1.01545, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00283, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00353, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.03448, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.05714, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.01363, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00741, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.02534, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.02284, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.29584, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.12825, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.08506, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0832, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.01904, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.65748, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.17227, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.12301, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.27751, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.42461, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.26913, FrameCnt: 0 State: 0
Finding PSS... Peak: 3.57099, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.06946, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00755, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.03238, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.03445, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.0338, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00297, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.00823, FrameCnt: 0 State: 0
Finding PSS... Peak: 1.61606, FrameCnt: 0 State: 0
[... 44 lines cut ...]
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
Finding PSS... Peak: 7.27225, FrameCnt: 0 State: 1
^CLTESniffer_Core: Exiting...
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/yuan/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/yuan/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------
Num  RNTI     Max Mod     Active   Success      SNR(dB)  DL-UL_delay(us)   Other_Info
--------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are max 16QAM table, 0 RNTIs are max 64QAM table, 0 RNTIs are max 256QAM, 0 RNTIs are Unknown 

Destroyed Phy
Skipped subframe: 0 / 221
nof_decoded_locations, nof_cce, nof_missed_cce, nof_subframes, nof_subframe_collisions_dw, nof_subframe_collisions_up, time, nof_locations
0, 0, 0, 0, 0, 0, 0.000000, 0
Skipped subframes: 0 (-nan%)
Saving MAC PCAP file
Deleted DL Sniffer core

I could not find the problem.

hdtuanss, 2023-06-30:
Have you changed the mtu of 10G card interface to 9000? I still see `[INFO] [X300] Maximum frame size: 1472 bytes.` in your log. That means the current mtu is 1472, could you use `ifconfig` command to find out the name of interface and execute `sudo ifconfig <10Gb card interface name> mtu 9000` again?

CumtZhangkuiyuan, 2023-06-30:
Thank you very much! I have changed  the mtu of 10G card interface to 9000. I see [INFO] [X300] Maximum frame size: 8000 bytes. in my log.  And the above problem have not been solved. Is this due to the MTU not reaching 9000 bytes？Also, have you seen this warning as follow:

[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.

hdtuanss, 2023-06-30:
I dont think the MTU = 8000 is the problem. The warning is also no problem too. Could you try to run srsRAN or FALCON with your USRP? Or please try downlink sniffing mode to see whether it works or not. I don't have any idea about this.

863356170, 2023-11-16:
Hello, I have a similar issue just like yours except I'm using bladerf. I got this while executing the downlink signal sniffing command: 
/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_blade_imp.c:449: RX failed: Operation timed out; nsamples=23040;
/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()
 Would you mind tell me if you have solved this problem and how?

863356170, 2023-11-16:
> Hello, I have a similar issue just like yours except I'm using bladerf. I got this while executing the downlink signal sniffing command:您好，我和您一样也有类似的问题，只是我使用的是 bladerf。我在执行下行链路信号嗅探命令时得到了这个： /LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_blade_imp.c:449: RX failed: Operation timed out; nsamples=23040;/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_blade_imp.c：449：RX 失败：操作超时;nsamples=23040; /LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c：772：接收样本时出错 /LTESniffer/src/src/LTESniffer_Core.cc:370: Error calling srsran_ue_sync_work()/LTESniffer/src/src/LTESniffer_Core.cc：370：调用 srsran_ue_sync_work（） 时出错 Would you mind tell me if you have solved this problem and how?你介意告诉我你是否已经解决了这个问题以及如何解决吗？

OK, I solved my problem now ... I just wrongly used USB2.0 and when I change to USB 3.0 it works. Another problem I met during this period was "only 2 ports supported" and it could be solved by using only one antenna. Just if someone meet it like me.
Sorry I stupidly interrupted this issue.

hdtuanss, 2023-11-16:
> OK, I solved my problem now ... I just wrongly used USB2.0 and when I change to USB 3.0 it works. Another problem I met during this period was "only 2 ports supported" and it could be solved by [...]

Thanks for sharing.

closed 2024-01-12 by hdtuanss