SysSec-KAIST/LTESniffer #29: Downlink sniffing for Security API ?

SpongeBobaa, 2023-06-29:
I saw in README.md that uplink sniffing is used to implement security API functionality. Can I use downlink sniffing to implement security API functionality?

hdtuanss, 2023-06-29:
Hi, the reason for using uplink sniffing mode for security API is most of the information (IMSI, TMSI, UECapabilityInformation) is taken from Uplink messages. If you use downlink sniffing mode, you will probably get information only from paging messages (TMSI and IMSI if you are lucky). Could you tell me which information you wanna obtain from security API?

SpongeBobaa, 2023-06-29:
Thank you for your answer. I want to find information about IMSI.
I looked at TMSI from the downlink analysis, but I haven't found any IMSI information yet.
i think it would be much better if I could check the UECapabilityInformation. I haven't analyzed the UECapabilityInformation yet。

hdtuanss, 2023-06-30:
If you wanna do research on IMSI using only downlink sniffing function, I suggest you start with srseNB and modify code of srseNB to generate IMSI paging. By doing that way, you can collect IMSI via downlink sniffing. Hope it is helpful for you.

closed 2023-07-13 by hdtuanss