SysSec-KAIST/LTESniffer #34: Problem about uplinklink sniffing

zjgsuzjx, 2023-07-13:
Hello, when I was executing the uplink signal sniffing command:
sudo ./src/LTESniffer -A 2 -W 2 -f 1850e6 -C -m 0
the following error occurred:
Thu_Jul_13_10.48.23_2023.
Creating Phy
Creating 2 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1301 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1316 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning receiver to 1850.000 MHz
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 436 FDD, CP: Normal  , DetectRatio=100% PSR=6.71, Power=-7.8 dBm
 Found Cell_id:  62 TDD, CP: Normal  , DetectRatio=100% PSR=2.18, Power=-21.5 dBm
Decoding PBCH for cell 436 (N_id_2=1)
Setting sampling rate 23.04 MHz
/LTESniffer/src/src/LTESniffer_Core.cc:228: Could not set sampling rate
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_1 specifies 1.92e+06, node 0/DDC_0 specifies 2.304e+07.
terminate called without an active exception

hdtuanss, 2023-07-13:
hi, could you modify your command to: `sudo ./src/LTESniffer -A 2 -W 2 -f 1850e6 -C -m 1` ?
it should be `-m 1` instead of `-m 0` for uplink.
Thanks

zjgsuzjx, 2023-07-13:
I‘m sorry, it's my mistake. When I run the downlink sniffering, it works fine with "-A 1," but when I use "-A 2," it throws the above error. Can you please advise on how to resolve this?

hdtuanss, 2023-07-13:
could you upgrade your UHD version to a version > 4.0.0 ?
Thanks

zjgsuzjx, 2023-07-13:
When I use version UHD 4.0.0, the following prompt appears and the program stops, but when I use UHD 3.15 with -A 1, it runs successfully.

> Creating Phy
> Creating 2 Worker threads 
> Opening RF device with 2 RX antennas...
> Available RF device list: UHD 
> Trying to open RF device 'UHD'
> [INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.0.0.HEAD-0-g90ce6062
> [INFO] [LOGGING] Fastpath logging disabled at runtime.
> Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
> [INFO] [UHD RF] RF UHD Generic instance constructed
> [INFO] [X300] X300 initialization sequence...
> [INFO] [X300] Maximum frame size: 1472 bytes.
> [INFO] [X300] Radio 1x clock: 184.32 MHz
> [WARNING] [UDP] The recv buffer could not be resized sufficiently.
> Target sock buff size: 24266666 bytes.
> Actual sock buff size: 212992 bytes.
> See the transport application notes on buffer resizing.
> Please run: sudo sysctl -w net.core.rmem_max=24266666
> [WARNING] [UDP] The send buffer could not be resized sufficiently.
> Target sock buff size: 24266666 bytes.
> Actual sock buff size: 212992 bytes.
> See the transport application notes on buffer resizing.
> Please run: sudo sysctl -w net.core.wmem_max=24266666
> [WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
> [WARNING] [UDP] The current send_buff_size of 212992 is less than the minimum recommended size of 307200 and may result in dropped packets on some NICs

hdtuanss, 2023-07-13:
could you execute following commands?
```
sudo sysctl -w net.core.wmem_max=24266666
sudo sysctl -w net.core.rmem_max=24266666
sudo ifconfig (your-10Gb-card-interface-name) mtu 9000
```

closed 2024-01-12 by hdtuanss