SysSec-KAIST/LTESniffer #40: How can I save DCI as a file? (How to use option -D)

DancingSW, 2023-10-18:
Hi,

I am attempting to use DCI information for traffic analysis. However, I've come across a couple of issues:

- When I use the option -D to save DCI information with the "DCIToFile" function, **the resulting file is empty**.
- Using the -d (debug) option, I am able to obtain brief DCI information.

While I'm comfortable using the "-d" option to collect DCI information, I'm curious about the correct usage of the "-D" option. Any advice or tips you can provide would be greatly appreciated.

Thank you so much for your contribution and hard work. This tool is invaluable for my research!

hdtuanss, 2023-11-02:
Hi, 
Sorry for my late reply!
To enable saving DCI information to file, please uncomment line #205 in SubframeWorker.cc
```
  // common.consumeDCICollection(subframeInfo); //save DCI to file
```
you will get DCI saved to file after that.
Thanks

DancingSW, 2023-11-21:
Thank you so much!

closed 2023-11-21 by DancingSW