SysSec-KAIST/LTESniffer #43: Is this compatible with B200 mini or B205 mini. I seem to get an error

wannerguzman, 2023-11-09:
sudo ./src/LTESniffer -A 2 -W 4 -f 1840e6 -u 1745e6 -C -m 1

LTESniffer

Wed_Nov__8_21.47.26_2023.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.6.0.0-0-g0c4637ab
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: num_recv_frames=512,recv_frame_size=8000,serial=32528BE,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200mini
[INFO] [B200] Loading FPGA image: /usr/local/share/uhd/images/usrp_b200mini_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
/home/charon/work/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:931: uhd_init failed, freeing...
[ERROR] [UHD RF] RuntimeError: Invalid option chosen for: clock source
Unable to open RF device 'UHD'
/home/charon/work/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error opening rf_a
^CLTESniffer_Core: Exiting...
^Z
[9]+  Stopped                 sudo ./src/LTESniffer -A 2 -W 4 -f 1840e6 -u 1745e6 -C -m 1

hdtuanss, 2023-11-09:
Hi, B200/B205 mini only has 1 RX antenna, please re-run the command with -A 1.
Thanks

wannerguzman, 2023-11-09:
I still get the same error. I am using the LTESniffer_multi_usrp branch. I'm I using the right branch?

hdtuanss, 2023-11-09:
If you are running UL sniffing mode with 2 B200s, please make sure that they are equipped with GPSDO and GPSDO is in locked mode.
If you are running DL sniffing mode with a single B200/B205, please modify the command:
```
sudo ./src/LTESniffer -A 1 -W 4 -f 1840e6 -u 1745e6 -C -m 0
```

wannerguzman, 2023-11-09:
I also just have an antenna connected to rx2, not sure if the others need to be connected?

hdtuanss, 2023-11-09:
> I still get the same error. I am using the LTESniffer_multi_usrp branch. I'm I using the right branch?

Could you check whether you have switched branch or not? And please also check the connection of 2 usrp, please run 
`uhd_find_devices` and update serial numbers to the code

wannerguzman, 2023-11-09:
Sorry I am a bit confused is it possible to run the software with just a single usrp b200 mini, If so what branch should I be on?

hdtuanss, 2023-11-09:
> Sorry I am a bit confused is it possible to run the software with just a single usrp b200 mini, If so what branch should I be on?

With a single USRP B200 mini, you can only run DL sniffing mode.
You need to use the `main` branch to use that mode.
If you wanna use UL sniffing mode, you need to have 2 USRP B200 or 1 USRP X310

closed 2023-11-09 by hdtuanss

stuitee, 2023-11-20:
I use an X310, but the following problem occurs. What is the reason? Thanks 
[image]

hdtuanss, 2023-11-20:
> I use an X310, but the following problem occurs. What is the reason? Thanks [image]

Hi,
Your USRP X310 is not connected to your PC successfully, please check your physical (cable, port) connection, as well as the UHD lib on your PC.
You can check the connection by: `uhd_find_device` and `uhd_usrp_probe`
Thanks.

wannerguzman, 2023-12-05:
> With a single USRP B200 mini, you can only run DL sniffing mode. You need to use the `main` branch to use that mode. If you wanna use UL sniffing mode, you need to have 2 USRP B200 or 1 USRP X310

> If you are running UL sniffing mode with 2 B200s, please make sure that they are equipped with GPSDO and GPSDO is in locked mode. If you are running DL sniffing mode with a single B200/B205, please modify the command:
> 
> ```
> sudo ./src/LTESniffer -A 1 -W 4 -f 1840e6 -u 1745e6 -C -m 0
> ```

When I perform this I get "Could not find any cell in this frequency"