SysSec-KAIST/LTESniffer #44: file mode

ZazCoder720, 2023-11-12:
Hello, I would like to inquire about the specific file format of the file mode. I cannot parse valid information using the file mode。

hdtuanss, 2023-11-14:
Hi,
The offline decoding function using I/Q recorded file is still in developing right now. It requires an additional code to record synchronized subframes to file which is used for the input of the offline decoding function.
I will update that function as soon as it is completed.

ZazCoder720, 2023-11-14:
> Hi, The offline decoding function using I/Q recorded file is still in developing right now. It requires an additional code to record synchronized subframes to file which is used for the input of the offline decoding function. I will update that function as soon as it is completed.

Thank you very much, but I have another question. When we use online mode, the hardware device we use is the USRP E312, but it seems that this device is not compatible with LTESniffer. If I want to use the USRP E312 device to run online mode, what modifications should I make? I hope you can help me. Thank you very much.

hdtuanss, 2023-11-16:
Hi, 
Are you using ethernet interface to connect USRP E312 and PC?
Could you please check the connection of your usrp by `uhd_find_device` and `uhd_usrp_probe` ?
If your usrp E312 is connected to PC correctly, you can use option `-a <usrp args>` to run LTESniffer to pass the USRP argument.
You can also run srsran with your usrp to check whether it works or not.
Thanks

ZazCoder720, 2023-11-16:
> 嗨，您是否使用以太网接口连接 USRP E312 和 PC？您能否通过 和 检查您的 usrp 的连接？如果您的 usrp E312 正确连接到 PC，您可以使用运行 LTESniffer 的选项来传递 USRP 参数。您还可以使用 usrp 运行 srsran 来检查它是否有效。 谢谢`uhd_find_device``uhd_usrp_probe``-a <usrp args>`

Okay, thank you for your suggestion. I will check one by one, and there is another question: Do you need a 10GB network card or above for data transmission support

Paloma-96, 2024-01-08:
Hi, I am writing to ask if there is any news regarding the add-on module needed to use the tool on capture files. 

Could you kindly explain what needs to be done to enable this functionality? How does one need to create the synchronized capture?

Thanks

cellular777, 2024-01-09:
> Hi, I am writing to ask if there is any news regarding the add-on module needed to use the tool on capture files.
> 
> Could you kindly explain what needs to be done to enable this functionality? How does one need to create the synchronized capture?
> 
> Thanks

@Paloma-96,  there was a similar functionality set up in FALCON. I can see parts of this in the current code. I think it requires re-enabling some of these components and testing them with the upgraded srsRAN and how LTESniffer is set up.

@hdtuanss, I have been off doing other things (I will send you a message). Are you still planning on file mode support in the short term? If not I may also look into adding this in the next month or so.

hdtuanss, 2024-01-12:
@Paloma-96 @cellular777 
Hi, actually I have already tested file mode (capturing subframes) several times. However, it seems to be unstable and easy to lose synchronization. I will try to release the first version this weekend (next weekend if it does not work well). After that, hopefully, you guys can test and help me improve the file mode.

@cellular777 After I release my code, let's see if we can merge them to improve its ability. Thanks for your contribution.

Paloma-96, 2024-01-12:
@hdtuanss 
Hi, thank you very much for the update. I am looking forward to be able to test the file mode.

hdtuanss, 2024-01-14:
> @hdtuanss Hi, thank you very much for the update. I am looking forward to be able to test the file mode.

HI, the file mode is updated, please test it
Thanks

closed 2024-01-19 by hdtuanss