SysSec-KAIST/LTESniffer #45: PRACH Signal Length Error

crest4162, 2023-12-16:
Firstly, thank you for making this project accessible.

I am using 2 USRP B205minis w/ a PPS for time alignment. I did some work (modifying srsRAN UHD APIs and the multi-usrp branch of LTESniffer) to enable the standard "wait for pps then set time on next pps" procedure. My setup works *partly* as expected. However, stderr is inundated with errors of the format:

```
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is X and should be Y
```

X and Y vary across cells but are always the exact same for any given cell.

However, among these errors are still successful PUSCH decodes.

Example snippet from run in uplink mode w/ security API `-z 3` and debug messages `-d`:
```
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
[A][PUSCH-16 ]  SF: 779 .3 -- RNTI: 36121  -- DL-UL(us): -2.35  -- SNR(db): 19.5   -- CQI RQ: 0|0 -- Noise Pwr: 41.4   -- MCS: 7   -- QPSK   -- SUCCESS -- Len: 13
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
786 -4    Contention Resolution     c1bfd3da         8865       RRC Connection Setup     
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
[A][PUSCH-64 ]  SF: 797 .3 -- RNTI: 36121  -- DL-UL(us): -2.39  -- SNR(db): 24.9   -- CQI RQ: 1|0 -- Noise Pwr: 37.4   -- MCS: 28  -- 64QAM  -- FAILED  -- Len: 469
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
[A][PUSCH-64 ]  SF: 798 .6 -- RNTI: 36121  -- DL-UL(us): -2.41  -- SNR(db): 23.7   -- CQI RQ: 1|0 -- Noise Pwr: 36.3   -- MCS: 28  -- 64QAM  -- FAILED  -- Len: 373
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
[A][PUSCH-16 ]  SF: 800 .4 -- RNTI: 12889  -- DL-UL(us): +3.69  -- SNR(db): 2.38   -- CQI RQ: 0|0 -- Noise Pwr: 19.2   -- MCS: 8   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-256]  SF: 800 .4 -- RNTI: 12889  -- DL-UL(us): +3.69  -- SNR(db): 2.38   -- CQI RQ: 0|0 -- Noise Pwr: 19.2   -- MCS: 8   -- 16QAM  -- FAILED  -- Len: 28
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
[A][PUSCH-16 ]  SF: 800 .7 -- RNTI: 12889  -- DL-UL(us): +6.38  -- SNR(db): 2.12   -- CQI RQ: 0|0 -- Noise Pwr: 17     -- MCS: 8   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-256]  SF: 800 .7 -- RNTI: 12889  -- DL-UL(us): +6.38  -- SNR(db): 2.12   -- CQI RQ: 0|0 -- Noise Pwr: 17     -- MCS: 8   -- 16QAM  -- FAILED  -- Len: 28
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
/root/LTESniffer-multiUSRP/build/srsRAN-src/lib/src/phy/phch/prach.c:982: srsran_prach_detect: Signal length is 4848 and should be 12288
```

Using the security API `-z 3`, I have never seen anything other than TMSIs from RRC Connection Setup messages. It makes sense that the absence of other identifiers is a consequence of these PRACH failures. 

I am not sure what to make of these errors. Looking around the `prach.c` file indicates that this has some relationship to the number of physical resource blocks. I can hack the code to silence these errors but that obviously doesn't get to the root of the problem. Does anyone have any guess as to what might be causing this?

hdtuanss, 2023-12-19:
Hi, thanks for your report.
I will examine this error on the weekend as right now I'm still working on another thing.
However, the PRACH error as you showed is not the reason why you didn't get any identifiers except TMSI; the PRACH itself does not contain any identifiers. Right now, please just ignore the error of PRACH.
To see other identifiers, please place your smartphone near LTESniffer and switch airplane mode (or turn off, turn on your phone). The identifier such as IMSI is only sent the first time your phone connects to the base station. Also, the uplink signal from the smartphone is significantly weak, so you might need to place it near the sniffer.
Thanks.

hdtuanss, 2023-12-20:
Hi,
Thanks for your investigation.

> I just now was looking at this error again and found something that might count as a fix. I referenced the [srsRAN eNodeB implementation](https://github.com/srsran/srsRAN_4G/blob/master/srsenb/src/phy/prach_worker.cc#L149) and noticed a missing term on line 688 of [UL_Sniffer_PUSCH.cc](https://github.com/SysSec-KAIST/LTESniffer/blob/LTESniffer-multi-usrp/src/src/UL_Sniffer_PUSCH.cc#L688) (multi-usrp branch). Changing this to
> 
> ```c++
> nof_sf * RSRAN_SF_LEN_PRB(enb_ul.cell.nof_prb) - prach.N_cp,
> ```
> 
> satisfies the check that was previously failing.

I think it is the exact reason for the error, I will try to use your code on LTESniffer later. Thanks for contribution.

> I have multiple UEs within a meter of the USRP antennas and I have played with different gain settings on the sniffer side w/o results. I have tried both toggling airplane mode and power cycling these UEs also. However, after applying the aforementioned fix to UL_Sniffer_PUSCH.cc and running in debug mode `-d`, I do see RACH messages exactly as expected when the UEs power on or exit airplane mode. Is there anything else that might be preventing any identifiers from being captured from uplinks?
> 
Could you please check api_collector.pcap and ltesniffer_ul_mode.pcap? Also, could you run LTESniffer, switch airplane mode on 1 smartphone, and share the terminal log (with option -d)? 
There are several additional reasons why the identifiers are not decoded:
+ The uplink packet that contains identifiers is divided into many small packets by UE (fragmentation). Consequently, LTESniffer cannot combine those small packets to decode identifiers. I have observed this thing on commercial networks several times
+ The UE uses a too-high modulation scheme (MCS index) on the uplink. This makes the uplink packet hard to decode as there are not many error-correcting bits inside the packet to facilitate the decoding process.

Please share the information as above and we can discuss more.
Thanks

crest4162, 2023-12-21:
@hdtuanss Hello again!

> Could you please check api_collector.pcap and ltesniffer_ul_mode.pcap?

Certainly. `api_collector.pcap` contains only `RRCConnectionSetup` messages. `ltesniffer_ul_mode.pcap` is primarily downlinks with the occaisional RACH and UL-SCH. I would be happy to share the files with you in a private manner. I am worried that the pcaps might contain sensitive information that shouldn't be published on this page.

> Also, could you run LTESniffer, switch airplane mode on 1 smartphone, and share the terminal log (with option -d)?

 [Here](https://github.com/SysSec-KAIST/LTESniffer/files/13734039/LTESniffer.txt) is the terminal log for a run with arguments `... -m 1 -z 3 -d`. The RACH is seen on line 93.
 
I'll also include another interesting test (see this [terminal log](https://github.com/SysSec-KAIST/LTESniffer/files/13734329/LTESniffer.txt)): I modified [this line](https://github.com/SysSec-KAIST/LTESniffer/blob/LTESniffer-multi-usrp/src/src/UL_Sniffer_PUSCH.cc#L266) to only show debug messages for successful decodes **or** SNR >= 10db. 

```cpp
if ((en_debug) && (pusch_res.crc || enb_ul.chest_res.snr_db >= 10))
```

After toggling airplane mode (RACH on line 21), I send pings to the network every 500ms. This phone stands out because of the large SNR and sudden jump in uplinks immediately after toggling airplane mode. However, still hardly any uplinks are decoded successfully. 

Let me know if it is helpful to have the pcaps and how I can best send them to you. Also I am happy to try some other tests if any come to mind. Thanks!

hdtuanss, 2023-12-21:
Hi, 
Could you send pcap files to my email: tuan.hoangdinh@kaist.ac.kr? I will examine all pcap files and the log you gave me.
Hope that we can figure out the problem and make it work.
Thanks.

crest4162, 2023-12-21:
@hdtuanss

> Hi, Could you send pcap files to my email: [tuan.hoangdinh@kaist.ac.kr](mailto:tuan.hoangdinh@kaist.ac.kr)? I will examine all pcap files and the log you gave me. Hope that we can figure out the problem and make it work. Thanks.

I just sent the files over. Thank you very much!

hdtuanss, 2023-12-22:
Hi, 
I checked the log and pcap files. There are several things I can see from them:
+ You are listening on a 4-port eNB, which is not supported by LTESniffer. This can lead to DCI missing and therefore some UL-DL packets are not detected and decoded.
```
 - Type:            FDD
 - PCI:             xxx
 **- Nof ports:       4**
 - CP:              Normal  
 - PRB:             100
 - PHICH Length:    Normal
 - PHICH Resources: 1/6
 - SFN:             84
```
+ After you switched Airplane mode, I dont see the RRC Connection Setup is decoded. I think it is missing due to 4-port problem as the first point. You can check my log below to see normal process:
```
[16:59:35] Processed 1000/1000 subframes
PRACH: 2/2, preamble=11, offset=71.5 us, peak2avg=274.3
[A][PUSCH-16 ]  SF: 687 .2 -- RNTI: 8869   -- DL-UL(us): -2.49  -- SNR(db): 20.4   -- CQI RQ: 0|0 -- Noise Pwr: 53.5   -- MCS: 5   -- QPSK   -- SUCCESS -- Len: 18 -- RAR
[A][PUSCH-16 ]  SF: 691 .5 -- RNTI: 8869   -- DL-UL(us): -2.47  -- SNR(db): 24.2   -- CQI RQ: 0|0 -- Noise Pwr: 47.4   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 22
[A][PUSCH-16 ]  SF: 693 .5 -- RNTI: 8869   -- DL-UL(us): -2.43  -- SNR(db): 23.7   -- CQI RQ: 1|0 -- Noise Pwr: 45.2   -- MCS: 8   -- QPSK   -- SUCCESS -- Len: 67
[A][PUSCH-16 ]  SF: 694 .2 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.6   -- CQI RQ: 1|0 -- Noise Pwr: 44.3   -- MCS: 8   -- QPSK   -- SUCCESS -- Len: 67
[A][PUSCH-16 ]  SF: 694 .3 -- RNTI: 8869   -- DL-UL(us): -2.41  -- SNR(db): 24.2   -- CQI RQ: 0|0 -- Noise Pwr: 43.7   -- MCS: 8   -- QPSK   -- SUCCESS -- Len: 67
[A][PUSCH-16 ]  SF: 694 .4 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.3   -- CQI RQ: 0|0 -- Noise Pwr: 44.2   -- MCS: 17  -- 16QAM  -- SUCCESS -- Len: 161
[A][PUSCH-16 ]  SF: 694 .5 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 24.1   -- CQI RQ: 1|0 -- Noise Pwr: 42.9   -- MCS: 5   -- QPSK   -- SUCCESS -- Len: 41
[A][PUSCH-16 ]  SF: 694 .6 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.9   -- CQI RQ: 0|0 -- Noise Pwr: 43.2   -- MCS: 5   -- QPSK   -- SUCCESS -- Len: 41
[A][PUSCH-16 ]  SF: 694 .7 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.3   -- CQI RQ: 0|0 -- Noise Pwr: 43     -- MCS: 6   -- QPSK   -- SUCCESS -- Len: 49
[A][PUSCH-16 ]  SF: 694 .8 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.3   -- CQI RQ: 1|0 -- Noise Pwr: 44.2   -- MCS: 19  -- 16QAM  -- SUCCESS -- Len: 193
[A][PUSCH-16 ]  SF: 694 .9 -- RNTI: 8869   -- DL-UL(us): -2.41  -- SNR(db): 23.7   -- CQI RQ: 0|0 -- Noise Pwr: 43.4   -- MCS: 11  -- 16QAM  -- SUCCESS -- Len: 85
[A][PUSCH-16 ]  SF: 695 .0 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.6   -- CQI RQ: 0|0 -- Noise Pwr: 43.5   -- MCS: 6   -- QPSK   -- SUCCESS -- Len: 49
[A][PUSCH-16 ]  SF: 695 .1 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 24.1   -- CQI RQ: 1|0 -- Noise Pwr: 41.4   -- MCS: 15  -- 16QAM  -- SUCCESS -- Len: 141
[A][PUSCH-16 ]  SF: 695 .3 -- RNTI: 8869   -- DL-UL(us): -2.41  -- SNR(db): 23.9   -- CQI RQ: 0|0 -- Noise Pwr: 42.9   -- MCS: 18  -- 16QAM  -- SUCCESS -- Len: 177
[A][PUSCH-16 ]  SF: 695 .4 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.6   -- CQI RQ: 1|0 -- Noise Pwr: 42.5   -- MCS: 18  -- 16QAM  -- SUCCESS -- Len: 177
[A][PUSCH-16 ]  SF: 695 .5 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 23.7   -- CQI RQ: 0|0 -- Noise Pwr: 42.1   -- MCS: 18  -- 16QAM  -- SUCCESS -- Len: 177
[A][PUSCH-16 ]  SF: 695 .6 -- RNTI: 8869   -- DL-UL(us): -2.42  -- SNR(db): 22.8   -- CQI RQ: 0|0 -- Noise Pwr: 42.7   -- MCS: 19  -- 16QAM  -- SUCCESS -- Len: 193
[A][PUSCH-16 ]  SF: 699 .6 -- RNTI: 8869   -- DL-UL(us): -2.44  -- SNR(db): 23.3   -- CQI RQ: 0|0 -- Noise Pwr: 43.2   -- MCS: 16  -- 16QAM  -- SUCCESS -- Len: 153
[A][PUSCH-16 ]  SF: 700 .3 -- RNTI: 8869   -- DL-UL(us): -2.45  -- SNR(db): 14.3   -- CQI RQ: 0|0 -- Noise Pwr: 36.1   -- MCS: 22  -- 64QAM  -- FAILED  -- Len: 173
[A][PUSCH-64 ]  SF: 700 .3 -- RNTI: 8869   -- DL-UL(us): -2.45  -- SNR(db): 14.3   -- CQI RQ: 0|0 -- Noise Pwr: 36.1   -- MCS: 22  -- 64QAM  -- SUCCESS -- Len: 173
[A][PUSCH-64 ]  SF: 703 .6 -- RNTI: 8869   -- DL-UL(us): -2.4   -- SNR(db): 23.6   -- CQI RQ: 0|0 -- Noise Pwr: 42     -- MCS: 22  -- 64QAM  -- SUCCESS -- Len: 233
[16:59:36] Processed 1000/1000 subframes
[16:59:37] Processed 1000/1000 subframes

```
+ There is no fragmentation of packets in your pcap files that might lead to missing identifiers.
 
At this point, could you find a 2-port eNB and run the experiment again?
Also, before running UL sniffing, please run downlink sniffing with option -d to check the SNR of the downlink signal
```
Unkown64 SF:334 -0 -- RNTI: 56772 -- DCI: 2A   -- Mod: QPSK   -- MCS: 1   -- nof_TB: 1 -- Byte: 18    -- Antenna SNR: 21.7|0    -- SNR: 22    -- Result:    FAILED
Unkown256SF:334 -0 -- RNTI: 56772 -- DCI: 2A   -- Mod: QPSK   -- MCS: 1   -- nof_TB: 1 -- Byte: 22    -- Antenna SNR: 21.7|0    -- SNR: 22    -- Result:    SUCCESS
256      SF:334 -3 -- RNTI: 61127 -- DCI: 2A   -- Mod: QPSK   -- MCS: 1   -- nof_TB: 1 -- Byte: 22    -- Antenna SNR: 19.1|0    -- SNR: 19.4  -- Result:    SUCCESS
Unkown64 SF:334 -3 -- RNTI: 38134 -- DCI: 2A   -- Mod: QPSK   -- MCS: 1   -- nof_TB: 1 -- Byte: 18    -- Antenna SNR: 19.1|0    -- SNR: 19.4  -- Result:    FAILED
Unkown256SF:334 -3 -- RNTI: 38134 -- DCI: 2A   -- Mod: QPSK   -- MCS: 1   -- nof_TB: 1 -- Byte: 22    -- Antenna SNR: 19.1|0    -- SNR: 19.4  -- Result:    SUCCESS
Unkown64 SF:334 -4 -- RNTI: 23402 -- DCI: 2A   -- Mod: QPSK   -- MCS: 0   -- nof_TB: 1 -- Byte: 11    -- Antenna SNR: 21  |0    -- SNR: 21.2  -- Result:    SUCCESS
```
Thanks.

crest4162, 2023-12-22:
@hdtuanss Ok great thank you very much for taking a look and for the recommendations. I will do some more testing and reply here soon.

For my own curiosity, why are 4-port eNBs not supported? I have struggled to understand the concept of antenna port so this might be a very basic question.

crest4162, 2024-01-08:
Ok here is an update after some more work:

Using the srsRAN cell search tool to examine cells in my area, I can see some non-4-port cells. However, LTESniffer does not manage to decode uplink traffic on these cells. This is probably because all of the UEs are far from the sniffer (all snr is < 3). So, I need to find a non-4-port eNB that my COTS phones will connect to. 

I installed some cell info apps (e.g., [Cell Tower Locator](https://play.google.com/store/apps/details?id=ru.v_a_v.celltowerlocator&pcampaignid=web_share)) on two different COTS phones and traveled to around my local area (dozens of km). I waited for the phones to connect to a new eNB then used srsRAN tools to further examine characteristics of the cells. To my surprise, I did not observe the phones connecting any non-4-port cell during these tests.

I was also surprised to find that the phones only reported two distinct channels used by this entire set of cells. I expected many more channels would be present to prevent interference across neighboring cells. [This Wikipedia page](https://en.wikipedia.org/wiki/Cellular_network#Concept) seems to reinforce my expectations. 

[image]

Unfortunately, I am kind of at a dead end now. I will happily repeat my tests to find a non-4-port eNB if someone has ideas for how to improve my methodology. 
 
@hdtuanss I would like to dedicate some effort towards this project to help enable 4-port sniffing, but would require direction as to what is preventing that functionality in the current version.

cellular777, 2024-01-09:
@crest4162, I would recommend opening up the entire project in something that you can search and browse the code if you have not. Maybe also have intellisense downloaded with the C-extension. VSCode has been helpful for me in navigating the code. 

I did track these types of errors down a while ago by going back from the error line number. There are limitations on srsRAN and the number of antennas the radio has for MIMO. I believe any MIMO configuration can be "supported" with any antenna combination, but it requires altering the algorithms in srsRAN. You could for example duplicate from 1x antenna the same receive data to the other ports (3D data). 

I thought about trying to do this but I am working on other components. It is important and would be a great feautre.

Here are some errors I get that are similar, mostly for DL packets.
- /home/LTESniffer/build/srsRAN-src/lib/src/phy/mimo/precoding.c:1853: Error predecoding multiplex: not implemented for 4 Tx ports
- /home/LTESniffer/build/srsRAN-src/lib/src/phy/phch/pdsch.c:880: Error predecoding
- /home/LTESniffer/build/srsRAN-src/lib/src/phy/mimo/precoding.c:1853: Error predecoding multiplex: not implemented for 4 Tx ports
- /home/LTESniffer/src/src/DL_Sniffer_PDSCH.cc:987: ERROR: Decoding PDSCH
- /home/LTESniffer/build/srsRAN-src/lib/src/phy/mimo/precoding.c:1219: Error predecoding CCD: Only 2 ports supported
- /home/LTESniffer/build/srsRAN-src/lib/src/phy/phch/sch.c:84: Invalid input 0 (min: 2, max: 15)

My network setup looks like this: (has 4-port
- SFN:                                 44 
- PHICH Resources:              1 
- PHICH Length:         Normal 
- PRB:                                100 
- CP:                           Normal 
- Nof ports:                          4 
- PCI:                                  53 
- Type:                             FDD

crest4162, 2024-01-09:
@cellular777 thanks for taking the time to respond.

The initial problem in this thread (the PRACH signal length error) appears to be solved with the change mentioned in [this comment](https://github.com/SysSec-KAIST/LTESniffer/issues/45#issuecomment-1863180323). I *believe* that was not related to antenna configurations.

Oddly, I haven't run into the errors you listed above. What hardware are you using? I am on a dual USRP B205mini setup w/ a PPS for time sync.

It sounds like the port limitation is a characteristic of srsRAN itself? I looked around some more and it appears that [srsenb only supports 1 or 2 ports](https://github.com/srsran/srsRAN_4G/blob/master/srsenb/src/enb_cfg_parser.cc#L1206). Or is it a characteristic of how the sniffer layers configure and use srsRAN functions?

hdtuanss, 2024-01-12:
> @cellular777 thanks for taking the time to respond.
> 
> The initial problem in this thread (the PRACH signal length error) appears to be solved with the change mentioned in [this comment](https://github.com/SysSec-KAIST/LTESniffer/issues/45#issuecomment-1863180323). I _believe_ that was not related to antenna configurations.
> 
> Oddly, I haven't run into the errors you listed above. What hardware are you using? I am on a dual USRP B205mini setup w/ a PPS for time sync.
> 
> I suppose you are right that the port limitation is a characteristic of srsRAN itself. I looked around some more and it appears that [srsenb only supports 1 or 2 ports](https://github.com/srsran/srsRAN_4G/blob/master/srsenb/src/enb_cfg_parser.cc#L1206). It's weird to me that LTESniffer works in both SISO and 2-port MIMO but this doesn't generalize to a 4-port MIMO case. I would expect it to be an all-or-nothing situation for MIMO.

Hi, sorry for my late reply.

LTESniffer doesn't support 4-port base stations because it relies on srsRAN lib, which also does not support decoding functions for 4-port base stations. The 4-port BS might employ DCI formats different from those currently supported in srsRAN and LTESniffer. Developing a new function to decode these additional DCI formats is necessary. Decoding downlink messages in this context also demands more complex functions, including variations in MIMO processing and channel estimation. Therefore, extending the srsRAN function is also needed.
Currently, I dont have a plan to support 4-port BS.  I will consider it once srsRAN releases a new update that includes support for 4-port base stations.

closed 2024-01-12 by hdtuanss