SysSec-KAIST/LTESniffer #46: USRP N310 with LTESniffer

purvajoshi1, 2023-12-19:
Dear KAIST team members,

Greetings of the day!

I am Purva Joshi, a PhD student. Currently, I am using LTESniffer for analyzing downlink and uplink traffic and I have captured downlink traffic using USRP B210. However, we want to move forward with USRP N310 for uplink and downlink. I am facing some issues to capture and implement LTESniffer with N310. You mentioned that, it is possible to use USRP X310 for both downlink and uplink.

I am writing those errors here:

- [ERROR] [UHD RF] std::bad_alloc
- Error timed out while receiving samples from UHD.
- Error calling srsran_ue_sync_work()
- Error receiving samples
- Error decoding MIB
- Could not decode PBCH from CELL ID
- Error searching for cell
- Could not find any cell in this frequency Cell not found after 8 trials. Trying again (Press Ctrl+C to exit).

Hope to hear back from you soon. Many thanks in advance.

Regards,
Purva Joshi

hdtuanss, 2023-12-19:
Hi Purva Joshi,
Many people had the std::bad_alloc error; but currently, I still dont know the exact reason for it. It seems to be a conflict between your CPU,  Ubuntu version, UHD version, and srsRAN lib. The only way to solve that problem is trying to set up LTESniffer on Ubuntu 18.04, with UHD version 4.0. Could you please set up LTESniffer in that environment and tell me the result again? I believe that most of the other errors will be solved too.
After that, we can discuss the cell search error.
Thanks

purvajoshi1, 2023-12-21:
Hello,
As you said, I started LTESniffer on Ubuntu 18.04, with UHD version 4.0. Now there is another error when I try uplink and downlink sniffer. It is as followed:

/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_utils.c:207: Could not find any cell in this frequency
Cell not found after 7 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
^CLTESniffer_Core: Exiting...
/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:1340: Error timed out while receiving samples from UHD.
/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_cell_search.c:317: Error calling srsran_ue_sync_work()
/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/ue/ue_cell_search.c:277: Error searching cell
/home/purva/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_utils.c:204: Error searching cell
/home/purva/LTESniffer/src/src/LTESniffer_Core.cc:199: Error searching for cell
terminate called without an active exception
Aborted

Kindly help me with this. Thanks in advance.

Regards,
Purva Joshi

hdtuanss, 2023-12-21:
Hi again,
The error messages mean there is no cell or the signal of the cell in your area is too weak. Are you locating the USRP and antennas indoors? Could you move them outdoors and try again? If you have an Android phone, please use Cellular-Z app to check the signal power of cells in your area.
If the problem persists, please give a comment here and we can discuss more.
Thanks

purvajoshi1, 2023-12-21:
Also, one more thing I would like to know, is it possible that I use B210 for downlink and N310 for uplink sniffing?

hdtuanss, 2023-12-21:
> Hello, As you said, I started LTESniffer on Ubuntu 18.04, with UHD version 4.0. Now there is another error when I try uplink and downlink sniffer. It is as followed: [...]

Could you share the whole terminal log including the command you have executed?

hdtuanss, 2023-12-21:
> Also, one more thing I would like to know, is it possible that I use B210 for downlink and N310 for uplink sniffing?

I haven't tested LTESniffer with N310 yet. Could you confirm whether it supports 2 different frequencies at the same time as the USRP X310? If so, I think using N310 alone is enough for uplink sniffing. If unlucky that it does not support 2 different freqs concurrently, please try your combination together as I dont have any N310 to test right now. I'm also curious about result.

hdtuanss, 2023-12-22:
Hi, could you contact me via email as before? I would like to have access to your Ubuntu machine and examine the error. Since I dont have USRP N310, I would like to see how it works with LTESniffer on your machine.
Thanks

purvajoshi1, 2023-12-22:
Hi Tuan,
I have sent you an email. I will be available for the next 2 hours. Kindly reach out as soon as possible. Thank you.

hdtuanss, 2024-01-14:
Update: I and Purva have worked together and solved this problem. No USRP N310 works. I will share the patched code later. Thanks

purvajoshi1, 2024-01-18:
Issue is solved. Finally works LTESniffer in UL and DL mode.

abbdulwahab86, 2024-02-22:
Hi @purvajoshi1 , @hdtuanss 
I have tested multiple environments for LTESniffer with ni-2955 (Fpga USRP N310)
Ubuntu 22.04, with UHD version 4.1.0.6
Ubuntu 18.04, with UHD version 4.1

But unable to find any cell in both UL or DL modes.
If you spare some time to access my Ubuntu machine.

purvajoshi1, 2024-02-22:
Hi Abdul,
I am using USRP N310 with Ubuntu 18.04. I will try definitely with Anydesk, Let me know your available time. Thank you.

abbdulwahab86, 2024-02-22:
Thanks @purvajoshi1 
Can you please share email address, for convenience of communicating available time

purvajoshi1, 2024-02-22:
purva.joshi@phd.unipi.it is my email address.

abbdulwahab86, 2024-02-26:
@hdtuanss 

Kindly help to resolve the issue.  Environments for LTESniffer with ni-2955 (Fpga USRP Mboard: X310)

butter-factory@butterfactory-OptiPlex-7050:~/ltes/build/src$ ./LTESniffer -A 2 -W 6 -f 1812.6e6 -C -m 0 -g 75

_**LTESniffer

Mon_Feb_26_15.15.46_2024.
Creating Phy
Creating 6 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.5.0.HEAD-0-g471af98f
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929b
[WARNING] [GPS] update_cache(): Malformed GPSDO string: LC_XO, Firmware Rev 0.929b
[INFO] [X300] Radio 1x clock: 184.32 MHz
[ERROR] [DBMGR] The daughterboard manager encountered a recoverable error in init.
Loading the "unknown" daughterboard implementations to continue.
The daughterboard cannot operate until this error is resolved.
ValueError: TwinRX clock rate 92160000.000000 is not a multiple of the pfd freq 12500000.000000.
[ERROR] [DBMGR] The daughterboard manager encountered a recoverable error in init.
Loading the "unknown" daughterboard implementations to continue.
The daughterboard cannot operate until this error is resolved.
ValueError: TwinRX clock rate 92160000.000000 is not a multiple of the pfd freq 12500000.000000.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened
Tunning receiver to 1812.600 MHz
/home/butter-factory/ltes/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:460: Sensor 'lo_locked` not found.
/home/butter-factory/ltes/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:503: Could not lock reference clock source. Sensor: lo_locked=false
Searching for cell...
/home/butter-factory/ltes/build/srsRAN-src/lib/src/phy/rf/rf_utils.c:207: Could not find any cell in this frequency
Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
Cell not found after 30 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
*Found Cell_id:   0 FDD, CP: Normal  , DetectRatio=100% PSR=2.32, Power=-80.4 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 0 (N_id_2=0)
Cell not found after 31 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...**_

hdtuanss, 2024-02-26:
> @hdtuanss
>
> Kindly help to resolve the issue. Environments for LTESniffer with ni-2955 (Fpga USRP Mboard: X310)
>
> butter-factory@butterfactory-OptiPlex-7050:~/ltes/build/src$ ./LTESniffer -A 2 -W 6 -f [...]

Hi, 
sorry for my late reply. Did Purva help you to modify the code to work with N310? If not, I will share a patch to you tomorrow.
Thanks

purvajoshi1, 2024-02-26:
@hdtuanss : Hi Tuan, I tried to solve but he has NI -2955 with (Fpga USRP Mboard: X310) and I tried with anydesk but if I am not wrong he configured as N310. I was not able to solve due to my hectic schedule. Maybe you can give a chance. Thanks.

hdtuanss, 2024-02-27:
@purvajoshi1 @abbdulwahab86  I'm also using NI-2955 with X310 Mboard, it should be programmed with X310 FPGA image. Which USRP image are you using?

abbdulwahab86, 2024-02-27:
@hdtuanss @purvajoshi1 
I used this command 
uhd_image_loader --args="type=x300,addr=192.168.40.2,fpga=HG"

abbdulwahab86, 2024-02-27:
> Hi, sorry for my late reply. Did Purva help you to modify the code to work with N310? If not, I will share a patch to you tomorrow. Thanks

@hdtuanss Can you please share the patch

hdtuanss, 2024-02-27:
> > Hi, sorry for my late reply. Did Purva help you to modify the code to work with N310? If not, I will share a patch to you tomorrow. Thanks
> 
> @hdtuanss Can you please share the patch

I just went through your log and found that you are using TwinRX daughter board on your USRP, it is correct?

abbdulwahab86, 2024-02-27:
@hdtuanss  Yes I am using TwinRx. I was able to resolve the issue partially. Basically first i set Master clock to 200MHz opposed to 184MHz set as default in code for x300 type boards. 

Then in the UHD init several Tx channel related things are set, commenting these out enabled running sniffer on TwinRx 
[image]

However the actual sampling rate I’m getting is 1.923Mhz opposed to the 1.920Mhz. 
Please see the attached snapshot. 

Can you please suggest what changes I need to make to run sniffer on 2 MHz to rate

hdtuanss, 2024-02-27:
You should set the Master clock to 184.32 Mhz as it should be divided by a positive integer number to meet the LTE standard sampling rate (1.92/3.84/7.68/15.36/23.04/30.72 Mhz). 

In addition, I didn't test LTESniffer with TwinRx, it has a different channel mapping than USRP N310 which I fixed for Purva. Supporting TwinRx may require more effort to modify code of LTESniffer and I need TwinRx to test. In this case, it is better if I can access your PC by Anydesk or Teamviewer to examine TwinRx. However, this week does not work for me as I have many other tasks. If you are okay with next week, please send me an email to tuan.hoangdinh@kaist.ac.kr.
Thanks.

abbdulwahab86, 2024-02-29:
Hi @hdtuanss and @purvajoshi1 ,

I checked the code with LTESniffer with Ettus USRP X310 and it started working but i cannot find anything in

Finding PSS... Peak: 4.08332, FrameCnt: 0 State: 1
 - Type:            FDD
 - PCI:             26
 - Nof ports:       2
 - CP:              Normal  
 - PRB:             75
 - PHICH Length:    Normal
 - PHICH Resources: 1
 - SFN:             912
Decoded MIB. SFN: 912, offset: 1
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
Num  RNTI     Table       Active   New TX   ReTX     Success  HARQ     Normal   W_MIMO   W_pinfor Other    
--------------------------------------------------------------------------------------------------------
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 0 RNTIs are Unknown 

Destroyed Phy
Skipped subframe: 0 / 175517
nof_decoded_locations, nof_cce, nof_missed_cce, nof_subframes, nof_subframe_collisions_dw, nof_subframe_collisions_up, time, nof_locations
1278, 98, 86, 24, 0, 0, 0.000000, 164
Skipped subframes: 0 (0%)
Saving MAC PCAP file
Deleted DL Sniffer core

abbdulwahab86, 2024-02-29:
**Issue resolved.** Thanks a lot for @hdtuanss @purvajoshi1

abbdulwahab86, 2024-02-29:
> The TwinRX daughterboard does not operate with the master clock rate of 184.32e6; operation with this master clock rate will result in UHD errors such as the following:
> [ERROR] [DBMGR] The daughterboard manager encountered a recoverable error in init.
> Loading the "unknown" daughterboard implementations to continue.
> The daughterboard cannot operate until this error is resolved.
> ValueError: TwinRX clock rate 92160000.000000 is not a multiple of the pfd freq 12500000.000000.

Ref: [](https://kb.ettus.com/TwinRX)

closed 2024-05-09 by hdtuanss