SysSec-KAIST/LTESniffer #47: Compilation Issue on ARM

MoPhD, 2023-12-19:
Please see "Make Error at CMakeLists.txt:332 (message):
  no SIMD instructions found" below. 

Is LTESniffer supported on ARM (specifically Apple M2 Pro running Ubuntu 22.04.3 LTS in a VM)?

mo@ubuntu:~/LTESniffer/build$ cmake ../ 
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build type not specified: defaulting to RelWithDebInfo.
-- CMNALIB LIBRARIES: CMNALIB_LIBRARY-NOTFOUND
-- Could NOT find CMNALIB (missing: CMNALIB_LIBRARIES) 
Download CMNALIB as a subproject and build it.
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/mo/LTESniffer/build/cmnalib-download
[ 11%] Performing update step for 'cmnalib'
[ 22%] No patch step for 'cmnalib'
[ 33%] No configure step for 'cmnalib'
[ 44%] No build step for 'cmnalib'
[ 55%] No install step for 'cmnalib'
[ 66%] No test step for 'cmnalib'
[ 77%] Completed 'cmnalib'
[100%] Built target cmnalib
CMake Deprecation Warning at build/cmnalib-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMNALIB_LIBRARY: cmnalib
CMNALIB_LIBRARIES: cmnalib
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (mkl) does
  not match the name of the calling package (MKL).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindMKL.cmake:44 (find_package_handle_standard_args)
  CMakeLists.txt:115 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find mkl (missing: MKL_LIBRARIES MKL_CORE MKL_ILP MKL_SEQ) 
-- FFTW3F LIBRARIES: /usr/lib/aarch64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/aarch64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (fftw3f)
  does not match the name of the calling package (FFTW3F).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindFFTW3F.cmake:41 (find_package_handle_standard_args)
  CMakeLists.txt:121 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- FFT_LIBRARIES: /usr/lib/aarch64-linux-gnu/libfftw3f.so
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
[... 159 lines cut ...]
-- static LibConfig path: /usr/lib/aarch64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/aarch64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/aarch64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/aarch64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/aarch64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/aarch64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/aarch64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
SRSRAN_LIBRARIES: srsran_commonsrsran_physrsran_asn1srsran_gtpusrsran_macsrsran_pdcpsrsran_rlcsrsran_rfsrsran_radio
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.35") found components: program_options system thread 
MARCH - FLAGS: native
CMake Error at CMakeLists.txt:332 (message):
  no SIMD instructions found

-- Configuring incomplete, errors occurred!
See also "/home/mo/LTESniffer/build/CMakeFiles/CMakeOutput.log".
See also "/home/mo/LTESniffer/build/CMakeFiles/CMakeError.log".

hdtuanss, 2023-12-20:
Hi, 
It seems the SIMD instructions technique on Apple CPU is not compatible with LTESniffer. This SIMD, which is often supported in Intel CPUs, is used to accelerate the turbo decoder of LTESniffer and srsRAN lib. 
Could you build LTESniffer on an Ubuntu + Intel CPU machine?
Thanks.

hdtuanss, 2023-12-22:
> @MoPhD I saw this post a couple days ago and just now by chance saw [this CMake option](https://github.com/SysSec-KAIST/LTESniffer/blob/main/CMakeLists.txt#L22) that looks like it might solve your issue.
> 
> When running your command `cmake ..`, add an additional flag like so:
> 
> ```
> cmake .. -DDISABLE_SIMD=on
> ```

Thanks for your answer.

@MoPhD Could you try this way?

MoPhD, 2023-12-25:
Thank you for the responses, running using cmake .. -DDISABLE_SIMD=on appears to resolve the error.

closed 2023-12-25 by MoPhD