SysSec-KAIST/LTESniffer #48: File based logging and DL/UL mode

cellular777, 2024-01-09:
@hdtuanss 
I have been adding some capabilities to a local version of the repo.
1. [DL/UL Mode] added a DL & UL mode "#2" that would parse both PDSCH and PUSCH, I pretty much just went through the DL (#0) and UL (#1) modes and made a new state that combined both (where it made sense). This seems to be working.
2. [File Logging] A great part of this project is there is a ton of information, but this all going to the console isn't practical. Right now, depending on the mode, certain things are sent to the console. However, the rest is lost. I made a thread-safe file logger to capture the information (listed below) in time-stamped files. This preserves the console output but lets you see all the other data. For example, running in API security mode, but still wanting to see UL tables. I have some predefines to turn off file loggings as well.
  - DL tables
  - UL tables
  - DL DCI messages
  - UL DCI messages
  - security API tables
  - stderr messages

The above required a lot of changes, are you interested in the above #1 or #2? If so, I will see if I can get permission to share. 

Thanks again for the great work on this project!

hdtuanss, 2024-01-12:
Hi Stevens, 

Thanks for your great effort and contribution to LTESniffer.

1. In the original UL sniffing mode of LTESniffer, it still can decode PDSCH but is limited to several DCI formats that are only used in TM1/2, where we can use only 1 RX antenna.  For decoding other downlink messages in TM3/4, 2 RX antennas are needed. If you enable both UL+DL modes concurrently, there is only 1 antenna can be used for DL, as another one is used for UL. Could you share more details on how you customize your code for both UL+DL and which capability it can support more? 
2. You are right, I have also recognized the logs you mentioned and wanted to add them to LTESniffer. However, I did not have enough time to develop more features and also I wanted to keep LTESniffer as a base code so that people could customize the code for their purposes. I'm interested in your contribution and appreciate your effort a lot.

For discussion and code sharing, I think we have 2 options:
+ You can give me a pull request
+ Or you can add me to your project and we can see together.

For convenience, could you add my Twitter account: @hdtuanss. We can use DM on Twitter for fast sharing and contracting. 

Thanks,
Tuan

Paloma-96, 2024-01-12:
@cellular777 
I would really like to test your code with the new features you mentioned. Could you share it in some way? Maybe as suggested by @hdtuanss via a pull request?

cellular777, 2024-01-25:
@hdtuanss @Paloma-96 

oof, my formatting got weird (fixed above).

Sorry for the delay. And the subsequent delay. 
1. I noticed that the MCSTracking tables weren't being updated/printed with UL mode for the DL. It is mostly around that. I tried to go where each UL mode feature was and make sure the new "mode" had things that would allow DL as well. I realize the DL is still used in the UL mode but I wanted to track what DL messages I could even with 1 antenna. (I think this makes sense, but I will leave it up to you to decide)

The code is in a local repo. I got permission to share back out, so I will need to set up a repo with this account and then will do one of the above. @hdtuanss do you prefer the pull request or adding to a project (whatever is best for you). 

@hdtuanss ha, I dont use twitter but I will make one so we can chat more directly.

@hdtuanss @Paloma-96 I unfortunatey was pressed for time and am out the following week. Come 2/5/2024 I will focus on getting the above sorted. Sorry again for the delay...

On other news, I do have a couple fixes for the UL rntimanager errors on the other issue. #37 I will post under there in general what I did. But those changes will likely just be included in process above.

hdtuanss, 2024-01-26:
> @hdtuanss @Paloma-96
>
> oof, my formatting got weird (fixed above).
>
> Sorry for the delay. And the subsequent delay.
>
> 1. I noticed that the MCSTracking tables weren't being updated/printed with UL mode [...]

Hi Stevens,

Thanks for your update. I have also seen your update on another issue #37.

For both issues, I think a pull request would be better for us. Could you please set up your repo and make a pull request for me? It would be great if you could include the fix for issue #37 together on your pull request.

Also, on point 1 we are discussing, since your new mode "2" can run independently from original modes "0" and "1", I think it will not have any problem if we add that mode to LTESniffer. So, let's make a separate new mode first; then, I will compare and see if I can combine your mode "2" and mode "1" into one.

Once again, I appreciate your contribution a lot. If you dont have Twitter account, please give me your any social account so I can add yours. 

Thank you.