SysSec-KAIST/LTESniffer #52: 2 x B200 mini-i "no gpsdo detected" error (no internal GPSDO, using external GPSDO)

MoPhD, 2024-01-17:
Does LTESniffer allow for external GPSDO 10MHz input signal fed into 1PPS input port of USRP B200-mini?
Or does it require an internal GPSDO (which the USRP B200-mini-i does not have)?

Ubuntu 22.04.3 LTS (NOT running in a VM)

13th gen i9 CPU, 24GB RAM, 230GB free disk space

I am feeding a 10MHz source signal from a GPS-synchronized GPSDO into a 10MHz distribution amplifier.

see: https://www.amazon.com/GOOZEEZOO-NEO-6M-Positioning-Disciplined-Oscillator/dp/B0CMD4NWJH
see: https://www.amazon.com/Distribution-Amplifier-Frequency-Standard-Output/dp/B0B715ZMND

Both USRP B200-mini-i devices are connected to the distribution amplifier via their 1PPS input port.
see: https://www.ettus.com/all-products/usrp-b200mini-i-2/
All prequisites installed, and compiled using multi usrp version with both serial numbers entered and recompiled (obfuscated below).

user@host:~/LTESniffer/build$ sudo ./src/LTESniffer -A 2 -W 4 -f 1840e6 -u 1745e6 -C -m 1
[sudo] password for user: 

LTESniffer

Wed_Jan_17_14.16.48_2024.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: num_recv_frames=512,recv_frame_size=8000,serial=3XXXXXX,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
/home/user/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:931: uhd_init failed, freeing...
[ERROR] [UHD RF] LookupError: KeyError: update_clock_source: gpsdo selected, but no gpsdo detected!
Unable to open RF device 'UHD'
/home/user/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error opening rf_a

MoPhD, 2024-01-17:
user@host:~/LTESniffer/build$ uhd_usrp_probe --args type=b200,serial=3XXXXXX
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B200
|   |   serial: 3XXXXXX
|   |   name: MyB200
|   |   product: 1
|   |   revision: 5
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B200 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B200 TX dual DAC
|   |   |   |   Gain Elements: None

user@host:~/LTESniffer/build$ uhd_usrp_probe --args type=b200,serial=3XXXXXX
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B200
|   |   serial: 3XXXXXX
|   |   name: MyB200
|   |   product: 1
|   |   revision: 5
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B200 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B200 TX dual DAC
|   |   |   |   Gain Elements: None

MoPhD, 2024-01-17:
Per the manual here: https://files.ettus.com/manual/page_sync.html
It looks like we need to set the source as "external":

External PPS and 10 MHz reference signals

Most USRPs have SMA connnectors on the front- or back-panel to provide these signals (10 MHz reference and PPS). These signals could be provided by an [OctoClock](https://files.ettus.com/manual/page_octoclock.html), an external/third-party GPSDO, a measurement device's reference outputs, or some other clock-generating device.

Connect these SMA connectors to the reference sources. In your software, select the external reference inputs as clock and time sources:
usrp->set_clock_source("external");
usrp->set_time_source("external");

hdtuanss, 2024-01-18:
Hi, 

Unfortunately, LTESniffer does not support external clock sources and time references right now. I tested the sniffer with PPS from Octoclock, but it seems unable to synchronize between uplink and downlink. Thus, I didn't implement that function anymore. 

Also, I took a look at the USRP B200-mini-i and saw that it does not support the GPSDO clock source as its motherboard cannot mount with the GPSDO module. Therefore, I worry LTESniffer will not work with B200-mini-i and you might need to use B200 (not mini) to make it work. 

If you still wanna try the external clock source, you can change the code in the file `LTESniffer_Core.cc` to **external** as follows:
```
  std::string rf_a_string = "clock=external,num_recv_frames=512,recv_frame_size=8000,serial=xxx"; 
  std::string rf_b_string = "clock=external,num_recv_frames=512,recv_frame_size=8000,serial=xxx";
``` 
However, based on my experiments before, I worry it will not work. Please share the result later.

Thanks.

MoPhD, 2024-01-18:
I made the change above, and ran make -j4. I receive the following output when running (I tried both 1PPS and 10MHz sources)
user@host:~/LTESniffer/build/src$ sudo ./LTESniffer -A 2 -W 4 -f 1840e6 -u 1745e6 -C -m 1
[sudo] password for user: 

LTESniffer

Wed_Jan_17_22.10.01_2024.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: num_recv_frames=512,recv_frame_size=8000,serial=3XXXXXX,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
/home/user/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:832: Could not lock reference clock source. Sensor: ref_locked=false

[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
/home/user/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:931: uhd_init failed, freeing...
[ERROR] [UHD RF] LookupError: IndexError: multi_usrp: RX channel 1 out of range for configured RX frontends
Unable to open RF device 'UHD'
/home/user/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error opening rf_a

hdtuanss, 2024-01-18:
Hi,

I'll look into this issue more thoroughly over the weekend. Currently, I can't assure you that it will work. In the meantime, if possible, please use the USRP B200 (not mini) to set up your experiment.

Thanks.

MoPhD, 2024-01-18:
Thank you, since I was only able to afford 2 x B200 mini, which together still cost over $1100 USD, I would appreciate if we could get this to work, I wouldnt be able to afford any more SDR equipment such as B200 (not mini) or B210 (not mini).

hdtuanss, 2024-01-21:
Hi,

Unfortunately, I tested with external PPS input but it doesn't seem to work. It is hard to achieve precise synchronization between UL and DL channels when using external PPS. I have tried several ways but still does not work. This may require more time to develop and test. Please switch to other SDRs that work right now, such as B200 or B210. I'll provide further updates as soon as possible.
Thanks

MoPhD, 2024-03-21:
I have acquired 2 x B200 with internal GPS-DO. I am getting the following error after recompiling and running:

me@mine:~/LTESniffer/build/src$ sudo ./LTESniffer -A 2 -W 4 -f 1840e6 -u 1745e6 -C -m 1
[sudo] password for me: 

LTESniffer

Thu_Mar_21_08.12.30_2024.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: num_recv_frames=512,recv_frame_size=8000,serial=8XXXXX1,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.933b Produced by HHHF
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
/home/me/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_uhd_imp.cc:931: uhd_init failed, freeing...
[ERROR] [UHD RF] LookupError: IndexError: multi_usrp: RX channel 1 out of range for configured RX frontends
Unable to open RF device 'UHD'
/home/me/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_imp.c:145: Failed to open a RF frontend device. Please check the available srsRAN CMAKE options to verify what RF frontend devices have been detected in your system
Error opening rf_a

hdtuanss, 2024-03-21:
Could you try again with -A 1?
sudo ./LTESniffer -A 1 -W 4 -f 1840e6 -u 1745e6 -C -m 1

MoPhD, 2024-03-22:
Thanks that worked!

MoPhD, 2024-03-22:
Thanks that worked!

closed 2024-03-22 by MoPhD