SysSec-KAIST/LTESniffer #54: Error

Y20G, 2024-02-05:
Hi all,

I'm running Dragonos and using HackRF One. However, while building LTESniffer, I'm getting this error. I think the error is related to USRP as I'm using HackRF One only. here is the error:-
Please help me in resolving----

Building C object srsRAN-build/lib/src/phy/phch/test/CMakeFiles/prach_test_usrp.dir/prach_test_usrp.c.o
[ 42%] Linking CXX executable prach_test_usrp
/usr/bin/ld: ../../rf/libsrsran_rf.so.21.10.0: undefined reference to `uhd::_log::log::log(uhd::log::severity_level, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::thread::id)'
collect2: error: ld returned 1 exit status

hdtuanss, 2024-02-07:
Hi, did you install full driver for HackRF One before building LTESniffer?
As far as I know, srsRAN lib does not support HackRF One. In addition, I dont have HackRF to test and optimize LTESniffer for it.
Therefore, I worry that I cannot help you much to solve this problem.

anoduck, 2024-03-09:
@Y20G You might want to checkout [this project](https://github.com/tomdionysus/SoapyHackRFDuplex) I just discovered.

closed 2024-05-09 by hdtuanss

andyraf, 2024-09-19:
I have the exact same problem trying to build under Dragon OS and opened a new ticket before finding this. Did anyone ever come up with a resolution?

anoduck, 2024-09-19:
@andyraf `collect2: error: ld returned 1 exit status` simply means that the build process failed during the linking stage [^1], and indicates an error occurred prior to this one. So, the real error is somewhere further up in your build output. 

If need be, you can place `make` in debug mode to generate additional output. `make -d A -vv`

One last tip, SDR projects don't receive as many updates as other system packages, so make sure the dependencies match.

[^1]: [What does 'collect2: error: ld...' mean?](https://stackoverflow.com/questions/27272525/what-does-collect2-error-ld-returned-1-exit-status-mean#27272651)