SysSec-KAIST/LTESniffer #55: Build Error: `no SIMD instructions found`

anoduck, 2024-02-15:
Installed all the requirements, went to build the application and recieved the error `no SIMD instructions found`. Which leads me to assume that either the code base is in the process of being refactored, I am missing a dependency, or my device is incompatible with the project.

```bash
Script started on 2024-02-15 04:50:46-05:00 [TERM="xterm-256color" TTY="/dev/pts/2" COLUMNS="272" LINES="68"]
Running with debug output on.
[0mCMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[0m
   Called from: [1]	/home/user/LTESniffer/CMakeLists.txt
-- Build type not specified: defaulting to RelWithDebInfo.
   Called from: [1]	/home/user/LTESniffer/CMakeLists.txt
-- CMNALIB LIBRARIES: CMNALIB_LIBRARY-NOTFOUND
   Called from: [2]	/home/user/LTESniffer/cmake/modules/FindCMNALIB.cmake
                [1]	/home/user/LTESniffer/CMakeLists.txt
-- Could NOT find CMNALIB (missing: CMNALIB_LIBRARIES) 
   Called from: [4]	/usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake
                [3]	/usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake
                [2]	/home/user/LTESniffer/cmake/modules/FindCMNALIB.cmake
                [1]	/home/user/LTESniffer/CMakeLists.txt
[33mCMake Warning (dev) at cmake/modules/FindCMNALIB.cmake:56 (MARK_AS_ADVANCED):
  Policy CMP0102 is not set: The variable named "CMNALIB_LIBRARIES" is not in
  the cache.  This results in an empty cache entry which is no longer created
  when policy CMP0102 is set to NEW.  Run "cmake --help-policy CMP0102" for
  policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.
Call Stack (most recent call first):
  CMakeLists.txt:72 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
[0m
   Called from: [2]	/home/user/LTESniffer/cmake/modules/FindCMNALIB.cmake
                [1]	/home/user/LTESniffer/CMakeLists.txt
[0mDownload CMNALIB as a subproject and build it.[0m
   Called from: [1]	/home/user/LTESniffer/CMakeLists.txt
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/user/LTESniffer/build/cmnalib-download
[ 11%] Performing update step for 'cmnalib'
[ 22%] No patch step for 'cmnalib'
[ 33%] No configure step for 'cmnalib'
[ 44%] No build step for 'cmnalib'
[ 55%] No install step for 'cmnalib'
[ 66%] No test step for 'cmnalib'
[ 77%] Completed 'cmnalib'
[100%] Built target cmnalib
[0m   Entering             /home/user/LTESniffer/build/cmnalib-src[0m
   Called from: [1]	/home/user/LTESniffer/CMakeLists.txt
[0mCMake Deprecation Warning at build/cmnalib-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[0m
   Called from: [1]	/home/user/LTESniffer/build/cmnalib-src/CMakeLists.txt
[0m   Entering             /home/user/LTESniffer/build/cmnalib-src/src[0m
   Called from: [1]	/home/user/LTESniffer/build/cmnalib-src/CMakeLists.txt
[0m   Entering             /home/user/LTESniffer/build/cmnalib-src/src/cmnalib[0m
[... 483 lines cut ...]
   Called from: [2]	/home/user/LTESniffer/cmake/modules/FindSSE.cmake
                [1]	/home/user/LTESniffer/build/srsRAN-src/CMakeLists.txt
[33mCMake Warning (dev) at cmake/modules/FindSSE.cmake:176 (mark_as_advanced):
  Policy CMP0102 is not set: The variable named "HAVE_FMA," is not in the
  cache.  This results in an empty cache entry which is no longer created
  when policy CMP0102 is set to NEW.  Run "cmake --help-policy CMP0102" for
  policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.
Call Stack (most recent call first):
  build/srsRAN-src/CMakeLists.txt:388 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
[0m
   Called from: [2]	/home/user/LTESniffer/cmake/modules/FindSSE.cmake
                [1]	/home/user/LTESniffer/build/srsRAN-src/CMakeLists.txt
[31mCMake Error at build/srsRAN-src/CMakeLists.txt:494 (message):
  no SIMD instructions found

[0m
   Called from: [1]	/home/user/LTESniffer/build/srsRAN-src/CMakeLists.txt
[0m   Returning to         /home/user/LTESniffer[0m
   Called from: [1]	/home/user/LTESniffer/CMakeLists.txt
-- Configuring incomplete, errors occurred!
Script done on 2024-02-15 04:51:22-05:00 [COMMAND_EXIT_CODE="1"]

```

hdtuanss, 2024-02-26:
Hi, sorry for my late reply, please refer to issue #47 to fix this issue.
thanks

anoduck, 2024-02-26:
@hdtuanss `cmake -DDISABLE_SIMD=on ..`, and installing some missing dependencies, got me a littler further. But, I am still encountering a lot of build errors.

libconfig++-dev & libusrsctp-dev were missing, both now recognized. I am also unable to find cpufreqtools in kali's repos, but I was able to find libcpupower-dev, which appears to do it. 

Still receiving a lot of build errors.  `field 'value' has incomplete type 'std::array<char, 256>'`

hdtuanss, 2024-02-27:
Could you try to use Ubuntu 18.04? I believe that I will solve all errors you are having.

anoduck, 2024-02-27:
@hdtuanss I might look into using docker or Vbox to emulate Ubuntu later on. For now, I have reviewed the cmake output, and discovered a few more dependencies not satisfied. Installed libdwarf-dev, mklibs, binutils-dev, libzmq3-dev, and libfduserdata-dev to be safe. 

The dependencies I am having difficulty satisfying are the `srsGui` and SKIQ requirements. SKIQ is not found in the Ubuntu package repos, and I have yet to find it in a github repo. `srsGui` fails to build at the moment due to the use of a deprecated header in the source. I have opened up an issue [there](https://github.com/srsran/srsGUI/issues/23). MbedTLS is satisfied, but the location of the library will need to be added for the next attempt at compilation. 

Since Kali is basically Debian sid, which is equivalent to Ubuntu-dev, it is unclear if changing operating system environments will provide any benefits towards overcoming the issue of the deprecated header reference in `srsGui`. Downgrading from a release to an Ubuntu LTS might though. So, it is worth considering.

anoduck, 2024-02-28:
@hdtuanss What about building it in Docker with ubuntu? Shouldn't that work just fine?

anoduck, 2024-02-29:
Beginning work on writing up a Dockerfile and docker-compose.yml for LTESniffer based off of Ubuntu 18.04. Repo will be at https://github.com/anoduck/LTESniffer-docker

closed 2024-02-29 by anoduck