SysSec-KAIST/LTESniffer #57: Uplink Decoding Performance

crest4162, 2024-03-04:
Hello again! For context, this post serves as a continuation of [a previous one](https://github.com/SysSec-KAIST/LTESniffer/issues/45) where I was attempting uplink decoding using a pair of USRP B205minis w/ a PPS for time alignment. That thread ended with a discussion about the limitation of the underlying srsRAN library to 1 and 2 port cells.

I have recently found a 2-port eNB (10 MHz bandwidth) that my COTS phones will connect to. I placed the sniffer in an area with good signal quality (>20 SNR). With a phone ~1 meter from the USRPs, I switched airplane mode on and off several times. Upon switching airplane mode off, the RACH and RRC Connection Setup can be seen, but no further results are present.

*The following text shows 5 instances of switching airplane mode from a run with arguments `... -m 1 -z 3 -d`. It was generated by filtering LTESniffer output with the command `grep -v Paging <sniffer output file> | grep -A 10 PRACH`.*

```
PRACH: 1/2, preamble=4, offset=69.6 us, peak2avg=276.2 
716 -2    Contention Resolution     1a0ff027         16547      RRC Connection Setup     
[A][PUSCH-16 ]  SF: 720 .5 -- RNTI: 16547  -- DL-UL(us): -1.25  -- SNR(db): 14.5   -- CQI RQ: 1|0 -- Noise Pwr: 37.5   -- MCS: 20  -- 16QAM  -- FAILED  -- Len: 161
[A][PUSCH-256]  SF: 720 .5 -- RNTI: 16547  -- DL-UL(us): -1.25  -- SNR(db): 14.5   -- CQI RQ: 1|0 -- Noise Pwr: 37.5   -- MCS: 20  -- 64QAM  -- FAILED  -- Len: 249
[A][PUSCH-16 ]  SF: 752 .6 -- RNTI: 16547  -- DL-UL(us): -12.3  -- SNR(db): 4.66   -- CQI RQ: 0|0 -- Noise Pwr: 46.4   -- MCS: 20  -- 16QAM  -- FAILED  -- Len: 105
[A][PUSCH-256]  SF: 752 .6 -- RNTI: 16547  -- DL-UL(us): -12.3  -- SNR(db): 4.66   -- CQI RQ: 0|0 -- Noise Pwr: 46.4   -- MCS: 20  -- 64QAM  -- FAILED  -- Len: 165
[A][PUSCH-16 ]  SF: 762 .6 -- RNTI: 16547  -- DL-UL(us): -1.16  -- SNR(db): 27.2   -- CQI RQ: 0|0 -- Noise Pwr: 28.9   -- MCS: 25  -- 64QAM  -- FAILED  -- Len: 141
[A][PUSCH-64 ]  SF: 762 .6 -- RNTI: 16547  -- DL-UL(us): -1.16  -- SNR(db): 27.2   -- CQI RQ: 0|0 -- Noise Pwr: 28.9   -- MCS: 25  -- 64QAM  -- SUCCESS -- Len: 141
776 -1    Contention Resolution     c6ecbf42         16555      RRC Connection Setup     
[A][PUSCH-16 ]  SF: 803 .7 -- RNTI: 6833   -- DL-UL(us): +0.859 -- SNR(db): 2.23   -- CQI RQ: 0|0 -- Noise Pwr: 15.2   -- MCS: 19  -- 16QAM  -- FAILED  -- Len: 97
[A][PUSCH-256]  SF: 803 .7 -- RNTI: 6833   -- DL-UL(us): +0.859 -- SNR(db): 2.23   -- CQI RQ: 0|0 -- Noise Pwr: 15.2   -- MCS: 19  -- 64QAM  -- FAILED  -- Len: 185
--
PRACH: 2/2, preamble=41, offset=28.6 us, peak2avg=343.0 
544 -1    Contention Resolution     f26e7dac         2181       RRC Connection Setup     
[A][PUSCH-16 ]  SF: 547 .4 -- RNTI: 2181   -- DL-UL(us): +9.94  -- SNR(db): 7.14   -- CQI RQ: 0|0 -- Noise Pwr: 49.7   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 9
[A][PUSCH-256]  SF: 547 .4 -- RNTI: 2181   -- DL-UL(us): +9.94  -- SNR(db): 7.14   -- CQI RQ: 0|0 -- Noise Pwr: 49.7   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-16 ]  SF: 547 .8 -- RNTI: 2181   -- DL-UL(us): -12.3  -- SNR(db): 4.55   -- CQI RQ: 0|0 -- Noise Pwr: 52     -- MCS: 2   -- QPSK   -- FAILED  -- Len: 9
[A][PUSCH-256]  SF: 547 .8 -- RNTI: 2181   -- DL-UL(us): -12.3  -- SNR(db): 4.55   -- CQI RQ: 0|0 -- Noise Pwr: 52     -- MCS: 2   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-16 ]  SF: 547 .9 -- RNTI: 2181   -- DL-UL(us): -12.3  -- SNR(db): 5.61   -- CQI RQ: 0|0 -- Noise Pwr: 52     -- MCS: 13  -- 16QAM  -- FAILED  -- Len: 141
[A][PUSCH-256]  SF: 547 .9 -- RNTI: 2181   -- DL-UL(us): -12.3  -- SNR(db): 5.61   -- CQI RQ: 0|0 -- Noise Pwr: 52     -- MCS: 13  -- 16QAM  -- FAILED  -- Len: 269
[A][PUSCH-16 ]  SF: 549 .4 -- RNTI: 2181   -- DL-UL(us): +9.95  -- SNR(db): 8.14   -- CQI RQ: 0|0 -- Noise Pwr: 49.4   -- MCS: 12  -- 16QAM  -- FAILED  -- Len: 97
[A][PUSCH-256]  SF: 549 .4 -- RNTI: 2181   -- DL-UL(us): +9.95  -- SNR(db): 8.14   -- CQI RQ: 0|0 -- Noise Pwr: 49.4   -- MCS: 12  -- 16QAM  -- FAILED  -- Len: 193
[A][PUSCH-16 ]  SF: 549 .4 -- RNTI: 18125  -- DL-UL(us): -0.598 -- SNR(db): 2.35   -- CQI RQ: 0|0 -- Noise Pwr: 17.5   -- MCS: 4   -- QPSK   -- FAILED  -- Len: 26
--
PRACH: 2/2, preamble=27, offset=35.3 us, peak2avg=253.2 
50  -1    Contention Resolution     3d6329c3         4557       RRC Connection Setup     
[A][PUSCH-16 ]  SF: 53  .1 -- RNTI: 4557   -- DL-UL(us): -1.3   -- SNR(db): 27.2   -- CQI RQ: 0|0 -- Noise Pwr: 26.6   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 53  .2 -- RNTI: 4557   -- DL-UL(us): -1.29  -- SNR(db): 25.5   -- CQI RQ: 0|0 -- Noise Pwr: 28.2   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 53  .3 -- RNTI: 4557   -- DL-UL(us): -1.33  -- SNR(db): 22.6   -- CQI RQ: 0|0 -- Noise Pwr: 31.1   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 53  .4 -- RNTI: 4557   -- DL-UL(us): -1.28  -- SNR(db): 27.3   -- CQI RQ: 0|0 -- Noise Pwr: 26.2   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 53  .9 -- RNTI: 4557   -- DL-UL(us): +9.81  -- SNR(db): 8.44   -- CQI RQ: 0|0 -- Noise Pwr: 45.6   -- MCS: 18  -- 16QAM  -- FAILED  -- Len: 177
[A][PUSCH-256]  SF: 53  .9 -- RNTI: 4557   -- DL-UL(us): +9.81  -- SNR(db): 8.44   -- CQI RQ: 0|0 -- Noise Pwr: 45.6   -- MCS: 18  -- 64QAM  -- FAILED  -- Len: 301
[A][PUSCH-16 ]  SF: 55  .0 -- RNTI: 4557   -- DL-UL(us): -1.3   -- SNR(db): 24.5   -- CQI RQ: 1|0 -- Noise Pwr: 28.9   -- MCS: 14  -- 16QAM  -- FAILED  -- Len: 125
[A][PUSCH-256]  SF: 55  .0 -- RNTI: 4557   -- DL-UL(us): -1.3   -- SNR(db): 24.5   -- CQI RQ: 1|0 -- Noise Pwr: 28.9   -- MCS: 14  -- 64QAM  -- FAILED  -- Len: 233
[A][PUSCH-16 ]  SF: 57  .0 -- RNTI: 4557   -- DL-UL(us): -1.29  -- SNR(db): 24.6   -- CQI RQ: 0|0 -- Noise Pwr: 30.4   -- MCS: 17  -- 16QAM  -- SUCCESS -- Len: 121
--
PRACH: 1/1, preamble=29, offset=78.2 us, peak2avg=267.7 
680 -1    Contention Resolution     e18faa11         7131       RRC Connection Setup     
[A][PUSCH-16 ]  SF: 682 .4 -- RNTI: 7131   -- DL-UL(us): -12.4  -- SNR(db): 4.32   -- CQI RQ: 0|0 -- Noise Pwr: 52.7   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 9
[A][PUSCH-256]  SF: 682 .4 -- RNTI: 7131   -- DL-UL(us): -12.4  -- SNR(db): 4.32   -- CQI RQ: 0|0 -- Noise Pwr: 52.7   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-16 ]  SF: 682 .8 -- RNTI: 7131   -- DL-UL(us): +9.82  -- SNR(db): 7.18   -- CQI RQ: 0|0 -- Noise Pwr: 49.6   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 9
[A][PUSCH-256]  SF: 682 .8 -- RNTI: 7131   -- DL-UL(us): +9.82  -- SNR(db): 7.18   -- CQI RQ: 0|0 -- Noise Pwr: 49.6   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-16 ]  SF: 682 .9 -- RNTI: 7131   -- DL-UL(us): -1.28  -- SNR(db): 25.4   -- CQI RQ: 0|0 -- Noise Pwr: 32.6   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 683 .2 -- RNTI: 7131   -- DL-UL(us): -1.27  -- SNR(db): 22.2   -- CQI RQ: 0|0 -- Noise Pwr: 35.3   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 9
[A][PUSCH-16 ]  SF: 684 .5 -- RNTI: 7131   -- DL-UL(us): -1.17  -- SNR(db): 14.2   -- CQI RQ: 0|0 -- Noise Pwr: 39.4   -- MCS: 2   -- QPSK   -- SUCCESS -- Len: 4
[A][PUSCH-16 ]  SF: 684 .6 -- RNTI: 7131   -- DL-UL(us): -12.4  -- SNR(db): 4.94   -- CQI RQ: 1|0 -- Noise Pwr: 51     -- MCS: 19  -- 16QAM  -- FAILED  -- Len: 145
[A][PUSCH-256]  SF: 684 .6 -- RNTI: 7131   -- DL-UL(us): -12.4  -- SNR(db): 4.94   -- CQI RQ: 1|0 -- Noise Pwr: 51     -- MCS: 19  -- 64QAM  -- FAILED  -- Len: 277
--
PRACH: 2/2, preamble=52, offset=21.0 us, peak2avg=150.1 
685 -1    Contention Resolution     4524531e         9365       RRC Connection Setup     
[A][PUSCH-16 ]  SF: 687 .2 -- RNTI: 9365   -- DL-UL(us): +9.81  -- SNR(db): 6.57   -- CQI RQ: 0|0 -- Noise Pwr: 44.9   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 9
[A][PUSCH-256]  SF: 687 .2 -- RNTI: 9365   -- DL-UL(us): +9.81  -- SNR(db): 6.57   -- CQI RQ: 0|0 -- Noise Pwr: 44.9   -- MCS: 2   -- QPSK   -- FAILED  -- Len: 15
[A][PUSCH-16 ]  SF: 689 .4 -- RNTI: 9365   -- DL-UL(us): +9.73  -- SNR(db): 7.86   -- CQI RQ: 1|0 -- Noise Pwr: 46.5   -- MCS: 14  -- 16QAM  -- FAILED  -- Len: 93
[A][PUSCH-256]  SF: 689 .4 -- RNTI: 9365   -- DL-UL(us): +9.73  -- SNR(db): 7.86   -- CQI RQ: 1|0 -- Noise Pwr: 46.5   -- MCS: 14  -- 64QAM  -- FAILED  -- Len: 173
[A][PUSCH-16 ]  SF: 691 .2 -- RNTI: 9365   -- DL-UL(us): -1.31  -- SNR(db): 8.07   -- CQI RQ: 0|0 -- Noise Pwr: 44.7   -- MCS: 20  -- 16QAM  -- FAILED  -- Len: 105
[A][PUSCH-256]  SF: 691 .2 -- RNTI: 9365   -- DL-UL(us): -1.31  -- SNR(db): 8.07   -- CQI RQ: 0|0 -- Noise Pwr: 44.7   -- MCS: 20  -- 64QAM  -- FAILED  -- Len: 165
[... cut ...]
[A][PUSCH-16 ]  SF: 719 .0 -- RNTI: 9365   -- DL-UL(us): -12.7  -- SNR(db): 2.91   -- CQI RQ: 0|0 -- Noise Pwr: 41.1   -- MCS: 19  -- 16QAM  -- FAILED  -- Len: 145
[A][PUSCH-256]  SF: 719 .0 -- RNTI: 9365   -- DL-UL(us): -12.7  -- SNR(db): 2.91   -- CQI RQ: 0|0 -- Noise Pwr: 41.1   -- MCS: 19  -- 64QAM  -- FAILED  -- Len: 277
``` 

I am not sure what to make of these results. I was surprised to see such variance in SNR for the same RNTI, and this appears to affect decoding reliability.

Thanks in advance!

hdtuanss, 2024-03-08:
Hi, could you share your experiment setup? I have not been successful in using PPS reference signal for synchronizing multiple USRPs. Did you change some code of LTESniffer to make it work? Based on my experiments before, using PPS seems to have bad sync.

crest4162, 2024-03-08:
Hi @hdtuanss,

I will happily share my setup.

I have two B205mini USRPs taking a PPS reference from a GPS evaluation kit (is just the easiest precision time pulse I have available).

I did make some code changes to LTESniffer:

### LTESniffer Diff 1
`git diff CMakeLists.txt`:
```sh
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 5ec2d0d..6afe713 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -188,6 +188,10 @@ else(SRSRAN_FOUND AND NOT FORCE_SUBPROJECT_SRSRAN)
   # let the compiler find included files from subdirectories
   include_directories(".")
 
+  file(COPY "${CMAKE_BINARY_DIR}/srsRAN-src/lib/src/phy/rf/"
+      DESTINATION "${CMAKE_BINARY_DIR}/srsRAN-src/lib/include/srsran/phy/rf"
+      FILES_MATCHING PATTERN "rf_uhd_*.h")
+
   # Add srsRAN directly to our build.
   add_subdirectory("${CMAKE_BINARY_DIR}/srsRAN-src"
     "${CMAKE_BINARY_DIR}/srsRAN-build")
```

### LTESniffer Diff 2
`git diff src/src/LTESniffer_Core.cc`:
```
diff --git a/src/src/LTESniffer_Core.cc b/src/src/LTESniffer_Core.cc
index 1f68cdb..089cfc9 100644
--- a/src/src/LTESniffer_Core.cc
+++ b/src/src/LTESniffer_Core.cc
@@ -28,6 +28,8 @@
 #undef I // Fix complex.h #define I nastiness when using C++
 #endif
 
+#include "srsran/phy/rf/rf_uhd_imp.h"
+
 #define ENABLE_AGC_DEFAULT
 using namespace std;
 
@@ -176,8 +178,8 @@ bool LTESniffer_Core::run(){
     printf("Opening RF device with %d RX antennas...\n", args.rf_nof_rx_ant);
     char rfArgsCStr_a[1024];
     char rfArgsCStr_b[1024];
-    std::string rf_a_string = "clock=gpsdo,num_recv_frames=512,recv_frame_size=8000,serial=3113D1B"; 
-    std::string rf_b_string = "clock=gpsdo,num_recv_frames=512,recv_frame_size=8000,serial=3125CB5";
+    std::string rf_a_string = "time_source=external,num_recv_frames=512,recv_frame_size=8000,serial=xxxxxxx";
+    std::string rf_b_string = "time_source=external,num_recv_frames=512,recv_frame_size=8000,serial=xxxxxxx";
 
     /*The following strings are for USRP X310 for specific application*/
     // std::string rf_a_string = "clock=gpsdo,type=x300,addr=192.168.40.2";
@@ -290,6 +292,16 @@ bool LTESniffer_Core::run(){
   }
 #endif
 
+  // usrp time sync via ref
+  printf("Synchronizing RF devices against PPS...\n");
+  void* handler_ptrs[2] = {rf_a.handler, rf_b.handler};
+  ret = rf_uhd_sync_pps_multi(handler_ptrs, 2);
+  if (ret != SRSRAN_SUCCESS) {
+    ERROR("Failed to synchronize RF devices: srsRAN errno %d\n", ret);
+    exit(-1);
+  }
+  printf("Synchronized RF devices.\n");
+
   /* If reading from file, go straight to PDSCH decoding. Otherwise, decode MIB first */
[... 135 lines cut ...]
 }
 #endif
```

### srsRAN Diff 4
`git diff lib/src/phy/rf/rf_uhd_safe.h`:
```
diff --git a/lib/src/phy/rf/rf_uhd_safe.h b/lib/src/phy/rf/rf_uhd_safe.h
index d09dfc7..bef44cc 100644
--- a/lib/src/phy/rf/rf_uhd_safe.h
+++ b/lib/src/phy/rf/rf_uhd_safe.h
@@ -145,6 +145,8 @@ public:
   virtual uhd_error get_sensor(const std::string& sensor_name, bool& sensor_value)       = 0;
   virtual uhd_error get_rx_sensor(const std::string& sensor_name, bool& sensor_value)    = 0;
   virtual uhd_error set_time_unknown_pps(const uhd::time_spec_t& timespec)               = 0;
+  virtual uhd_error set_time_next_pps(const uhd::time_spec_t& timespec)                  = 0;
+  virtual uhd_error get_time_last_pps(uhd::time_spec_t& timespec)                        = 0;
   virtual uhd_error get_time_now(uhd::time_spec_t& timespec)                             = 0;
   uhd_error         start_rx_stream(double delay)
   {
```

My changes to your multi-usrp fork of srsRAN kind of break API boundaries so some other hacks (a copy directive in CMakeLists.txt and an include in LTESniffer_core.cc - seen in LTESniffer diffs 1 and 2 above) are required to compile. For this same reason, I decided not to share these changes to GitHub.

I would be happy to push branches with these code changes if that is desired.

hdtuanss, 2024-03-11:
Hi, 
it seems you are having the same test as me before. I also tried to modify srsRAN code related to USRP to achieve synchronization by PPS. Unfortunately, the synchronization was really bad and that is the reason why you have many failed cases and variance in SNR. I still do not have any idea how to solve this problem. I hope you can solve it and definitely it will be a great contribution

crest4162, 2024-03-11:
Hi @hdtuanss thanks for getting back.

Could you please share info about your setup when you attempted PPS synchronization? More specifically, what USRPs did you attempt with? I am curious if you had similar issues with a different model of USRP.

crest4162, 2024-03-11:
After thinking about this problem more, I decided it might be useful to provide some additional information.

With the aforementioned PPS-synchronized dual-USRP setup, I have attempted some different eNB/UE combinations. This table shows PUSCH decoding reliability of LTESniffer with each combination of eNB and UE implementation. The sniffer was placed close to the eNB in each case to mitigate propagation delay.

<table>
  <tr>
    <td></td>
    <th>srsRAN eNB</th>
    <th>Commercial eNB</th>
  </tr>
  <tr>
    <th>srsRAN UE</th>
    <td>high success</td>
    <td>low success</td>
  </tr>
  <tr>
    <th>COTS UE</th>
    <td>high success</td>
    <td>low success</td>
  </tr>
</table>

I believe this is a partial validation of the PPS-synchronized setup.

crest4162, 2024-03-18:
Hi @hdtuanss thanks for getting back again.

This project is of high importance to me. I am in the position to offer compensation in return for improvements to decoding reliability and 4-port cell support in general. I will send you an email to continue this discussion.

If anyone else reading this is interested in the contract offer, please email me at crest4162@gmail.com.