SysSec-KAIST/LTESniffer #59: Offline Files

vvv123321, 2024-03-11:
Hello, I am very interested in your project, but I currently do not have any USRP equipment. Now I want to try running it using the recorded file. Can you send me the offline file that you have successfully tested? Thank you very much!
Additionally, I would like to ask, can the files generated by the "LTE Waveform Generator" tool in MATLAB 2019a be applicable to this project?
Looking forward to your reply!

vvv123321, 2024-03-12:
I hope it won't involve your private work. I would be very sorry if I offend you.

hdtuanss, 2024-03-14:
Hi, I will give you I/Q files this weekend. It will contain the traffic between srseNB and a commercial UE.
Regarding the MATLAB, as I'm not familiar with MATLAB, I'm sorry that I cannot help you on this problem.
Thanks

vvv123321, 2024-03-15:
@hdtuanss Okay, thank you very much for your reply. Once I receive the I/Q file you sent, I will try offline decoding work.
Looking forward to your reply again! Thank you!

hdtuanss, 2024-03-17:
Hi,
please download IQ sample file and the log in this link: https://drive.google.com/drive/folders/1AcEbU0JhV4uWIAVaYCf6B1M44zEPD4tW?usp=drive_link
you can refer to instruction here (https://github.com/SysSec-KAIST/LTESniffer/tree/LTESniffer-record-subframe) to decode recorded file.
Thanks

closed 2024-03-17 by hdtuanss

vvv123321, 2024-03-17:
Hello, I am glad to receive the file you sent. But it seems that there is only the log file "record_log" in the link, and there is no record of the recording file "subframe_iq_sample. bin" that receives subframe data.
Because I don't have USRP hardware on my end to record subframe data to the file "subframew_iq_sample. bin" using the command "sudo.//src/LTESniffer - A- f- C - m 0- n". I don't know if I misunderstood. I'm sorry to bother you again.

hdtuanss, 2024-03-17:
Hi,
sorry the previous drive was full, please use this one instead: https://drive.google.com/drive/folders/1KHUxzj-e55GhHBsIGz4oL98RNlsLau3j?usp=sharing
thanks