SysSec-KAIST/LTESniffer #60: Why can't I analyze the base station with a bandwidth of 15m?

stuitee, 2024-03-20:
When the bandwidth of the base station is 15M, it cannot be synchronized, obtaining MIB messages and all subsequent content.

hdtuanss, 2024-03-21:
Hi, please share your terminal log and hardware setup
Thanks

stuitee, 2024-03-21:
[image]
I am using USRP B210. When the bandwidth is 15M, I set parameter - A 1 and the program runs normally. However, when I set parameter A to 2, the program begins to show the results shown in the figure. Under other bandwidths, the setting of parameter A does not affect program operation

hdtuanss, 2024-03-21:
I hope you can copy and paste the whole terminal log including your executed command. By looking at your picture, is looks like there is no cell in your frequency or the signal power of this cell is too low

stuitee, 2024-03-22:
root@hmily-ThinkPad-P16-Gen-2:/home/hmily/LTESniffer/build/src# ./LTESniffer -A 1 -W 8 -g 76 -f 1870e6 -C -m 0 

LTESniffer 
Fri_Mar_22_08.59.22_2024.

Creating Phy

Creating 8 Worker threads

Opening RF device with 1 RX antennas...

Available RF device list: UHD

Trying to open RF device 'UHD'

[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa

[INFO] [LOGGING] Fastpath logging disabled at runtime.

Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6

[INFO] [UHD RF] RF UHD Generic instance constructed

[INFO] [B200] Detected Device: B210

[INFO] [B200] Operating over USB 3.

[INFO] [B200] Initialize CODEC control...

[INFO] [B200] Initialize Radio control...

[INFO] [B200] Performing register loopback test...

[INFO] [B200] Register loopback test passed

[INFO] [B200] Performing register loopback test...

[INFO] [B200] Register loopback test passed

[INFO] [B200] Asking for clock rate 23.040000 MHz...

[INFO] [B200] Actually got clock rate 23.040000 MHz.

RF device 'UHD' successfully opened

Tunning receiver to 1870.000 MHz

Searching for cell...

/home/hmily/LTESniffer/build/srsRAN-src/lib/src/phy/rf/rf_utils.c:207: Could not find any cell in this frequency

Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=83% PSR=2.18, Power=55.5 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Cell not found after 1 trials. Trying again (Press Ctrl+C to exit)

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=100% PSR=4.23, Power=62.0 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Setting sampling rate 15.36 MHz

[INFO] [B200] Asking for clock rate 61.440000 MHz...

[INFO] [B200] Actually got clock rate 61.440000 MHz.

Finding PSS... Peak: 1.03286, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.06344, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.22075, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.07674, FrameCnt: 0 State: 0

Finding PSS... Peak: 7.67707, FrameCnt: 0 State: 1

- Type:            FDD

- PCI:             1

- Nof ports:       1

- CP:              Normal 

- PRB:             75

- PHICH Length:    Normal

- PHICH Resources: 1/6

- SFN:             124

Decoded MIB. SFN: 124, offset: 1

[08:59:29] Processed 1000/1000 subframes

[08:59:30] Processed 1000/1000 subframes

[08:59:31] Processed 1000/1000 subframes

[08:59:32] Processed 1000/1000 subframes

Finding PSS... Peak: 1.00605, FrameCnt: 948 State: 0

Finding PSS... Peak: 7.87689, FrameCnt: 0 State: 1

- Type:            FDD

- PCI:             1

- Nof ports:       1

- CP:              Normal 

- PRB:             75

- PHICH Length:    Normal

- PHICH Resources: 1/6

- SFN:             604

Decoded MIB. SFN: 604, offset: 1

[08:59:33] Processed 1000/1000 subframes

[08:59:34] Processed 1000/1000 subframes

^CLTESniffer_Core: Exiting...

-------------------------------------------------------------------------------------------------------------

Num  RNTI     Table       Active   New TX   ReTX     Success      HARQ     Normal   W_MIMO   W_pinfor Other   

-------------------------------------------------------------------------------------------------------------

-------------------------------------------------------------------------------------------------------------

Num  RNTI     Table       Active   New TX   ReTX     Success      HARQ     Normal   W_MIMO   W_pinfor Other   

1    75       Unknown     591      591      0        109(18%)     0        0        0        0        0       

-------------------------------------------------------------------------------------------------------------

[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 1 RNTIs are Unknown  
Destroyed Phy

Skipped subframe: 0 / 6949

nof_decoded_locations, nof_cce, nof_missed_cce, nof_subframes, nof_subframe_collisions_dw, nof_subframe_collisions_up, time, nof_locations

4380, 81240, 8, 5416, 0, 0, 0.000000, 140816

Skipped subframes: 0 (0%)

Saving MAC PCAP file

Deleted DL Sniffer core

stuitee, 2024-03-22:
root@hmily-ThinkPad-P16-Gen-2:/home/hmily/LTESniffer/build/src# ./LTESniffer -A 2 -W 8 -g 76 -f 1870e6 -C -m 0 

LTESniffer 
Fri_Mar_22_08.59.43_2024.

Creating Phy

Creating 8 Worker threads

Opening RF device with 2 RX antennas...

Available RF device list: UHD

Trying to open RF device 'UHD'

[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-3-g080b1baa

[INFO] [LOGGING] Fastpath logging disabled at runtime.

Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6

[INFO] [UHD RF] RF UHD Generic instance constructed

[INFO] [B200] Detected Device: B210

[INFO] [B200] Operating over USB 3.

[INFO] [B200] Initialize CODEC control...

[INFO] [B200] Initialize Radio control...

[INFO] [B200] Performing register loopback test...

[INFO] [B200] Register loopback test passed

[INFO] [B200] Performing register loopback test...

[INFO] [B200] Register loopback test passed

[INFO] [B200] Asking for clock rate 23.040000 MHz...

[INFO] [B200] Actually got clock rate 23.040000 MHz.

[INFO] [MULTI_USRP]     1) catch time transition at pps edge

[INFO] [MULTI_USRP]     2) set times next pps (synchronously)

RF device 'UHD' successfully opened

Tunning receiver to 1870.000 MHz

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=88% PSR=2.44, Power=56.2 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=80% PSR=2.05, Power=55.5 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Cell not found after 1 trials. Trying again (Press Ctrl+C to exit)

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 TDD, CP: Normal  , DetectRatio=80% PSR=2.20, Power=56.3 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Cell not found after 2 trials. Trying again (Press Ctrl+C to exit)

Searching for cell...

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=100% PSR=4.83, Power=64.4 dBm

Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm

Decoding PBCH for cell 1 (N_id_2=1)

Setting sampling rate 15.36 MHz

[INFO] [B200] Asking for clock rate 61.440000 MHz...

[ERROR] [UHD RF] ValueError: current master clock rate (61.440000 MHz) exceeds maximum possible master clock rate (30.720000 MHz) when using 2 RX channels

[WARNING] [MULTI_USRP] The hardware does not support the requested RX sample rate:

Target sample rate: 15.360000 MSps

Actual sample rate: 11.520000 MSps 

[WARNING] [MULTI_USRP] The hardware does not support the requested RX sample rate:

Target sample rate: 15.360000 MSps

Actual sample rate: 11.520000 MSps 

Finding PSS... Peak: 1.64778, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.47886, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.05018, FrameCnt: 0 State: 0

Finding PSS... Peak: 1.22536, FrameCnt: 0 State: 0

hdtuanss, 2024-04-05:
Hi, the problem seems to be the sampling rate on USRP was not correctly handled. 
"[ERROR] [UHD RF] ValueError: current master clock rate (61.440000 MHz) exceeds maximum possible master clock rate (30.720000 MHz) when using 2 RX channels"
Could you try another USRP?

stuitee, 2024-04-07:
I tested with multiple USRP B210's separately and they all gave the same error.

hdtuanss, 2024-04-08:
Could you please modify the following code in "LTESniffer/build/srsRAN-src/lib/src/phy/common/phy_common.c" and build the project again?
[image]

hdtuanss, 2024-04-25:
> I tested with multiple USRP B210's separately and they all gave the same error.

Hi, any update on this problem?

stuitee, 2024-06-03:
#define FORCE_STANDARD_RATE

When I change the above code, the program doesn't work anymore.

DancingSW, 2024-10-31:
Hi all,

I have the same problem. I made a little bit of progress and want to share it.

- eNB in my area also has 15Mhz BW.
- Band is B66.
- eNB uses MIMO 4 TX and 4 RX.

I could get DCI at least by changing one related to **USE_LTE_RATES** line in CMakeLists.txt as follows:
[image]

LTESniffer does not work perfectly with the eNB in my area. I think that is because srsRAN does not support MIMO with 4 TX antennas. But I am fine with that cause what I need is DCI. Hope this helps for you.
[image]