SysSec-KAIST/LTESniffer #62: Download in offline mode.

ZazCoder720, 2024-03-30:
Hello, could you provide us with the offline mode download test file and debugging logs? We need this file to assist our project. If it's not very convenient, I'm sorry to disturb you.

hdtuanss, 2024-04-02:
Hi, please refer to issue #59. There is a link to download a test file.
Thanks

ZazCoder720, 2024-04-02:
Thank you very much, but there may have been some errors in my expression. What I meant was the testing files and debugging logs for the downlink, and that link is for the uplink. I'm sorry to trouble you again due to my expression error.

hdtuanss, 2024-04-05:
> Thank you very much, but there may have been some errors in my expression. What I meant was the testing files and debugging logs for the downlink, and that link is for the uplink. I'm sorry to trouble you again due to my expression error.

Hi, I will record an I/Q raw data file in downlink mode for you on this weekend. You can run LTESniffer using that file and see the DL debug messages.
Thanks

hdtuanss, 2024-04-08:
Hi,
Please download the I/Q raw data files on this link: 
[I/Q raw data files](https://drive.google.com/drive/folders/1KHUxzj-e55GhHBsIGz4oL98RNlsLau3j?usp=drive_link)
you can find the command to decode data in the log files.
Thanks

closed 2024-04-10 by hdtuanss