SysSec-KAIST/LTESniffer #65: cmake ../ ERROR

ad1tya-ahuja, 2024-04-10:
While configuring LTESniffer from the commands given in the repo, I get this error when running cmake ../
Using Ubuntu 20.04 and have installed all the dependencies.

ERROR:
[image]

purvajoshi1, 2024-04-12:
@ad1tya-ahuja: You can try using ubuntu 18.04, I tried and my few problems were solved.

ad1tya-ahuja, 2024-04-15:
> @ad1tya-ahuja: You can try using ubuntu 18.04, I tried and my few problems were solved.

[image]
Trying to run ubuntu 18.04 arm edition and this error pops up.

purvajoshi1, 2024-04-15:
@ad1tya-ahuja: I am not sure about this, but if you have a desktop where you can install Ubuntu 18.04 and follow step-by-step procedures you may solve problems related to CMAKE files. or else try to install cmake files with another version.

ad1tya-ahuja, 2024-04-15:
> Reference

[image]
Tried ubuntu 18.04 and got this error while running "make -j 4" command.

purvajoshi1, 2024-04-15:
@ad1tya-ahuja: The error indicates that your RF device is not able to open or not able to detect the RF device. If you are using BladeRF, make sure it is detected or you can share your screenshot. Hope you can solve the error.

hdtuanss, 2024-04-25:
Hi, 
sorry for my late response. The error you got is because your CPU does not support SIMD instructions which are used for accelerating physical layer decoders.
You might need to build LTESniffer on another CPU.
Or you can refer to issue #47 to disable SIMD. However, this can lead to performance issues.
Thanks