SysSec-KAIST/LTESniffer #66: Test file Error

vvv123321, 2024-04-15:
Hello, I am trying to generate an uplink test file using MATLAB tools and run it in LTESniffer offline mode. I found that in the LTESniffer_core.cc-Main loop code, n=0, the MIB could not be decoded and the correct information could not be printed (as shown in Figure 1). The terminal result is shown in Figure 2, and Figure 3 is the generated MATLAB test file (QPSK or QAM have both been tested, but neither of them works). This may be a problem with the generated file format or compatibility, how should I solve it? I would like to inquire about your opinions or ideas. Thank you!
[image]
[image]
[image]

hdtuanss, 2024-04-25:
Hi, I'm not familiar with Matlab but the MIB must be located in the first subframe in your generated file. Also, you might need to check whether LTESniffer is using standard sampling rate or 3/4 sampling rate by looking at the source file "LTESniffer/build/srsRAN-src/lib/src/phy/common/phy_common.c". 
[image]
If FORCE_STANDARD_RATE is defined, LTESniffer is gonna use standard rates (30.72 MHz for 20Mhz channel and so on). Otherwise, it is gonna use 3/4 sampling rate (23.04 Mhz for 20Mhz channel and so on)
Thanks

vvv123321, 2024-04-26:
Okay, thank you very much for your reply. I will review it according to your suggestions.

closed 2024-04-26 by vvv123321