SysSec-KAIST/LTESniffer #67: unable to build with system srsran

blshkv, 2024-04-15:
Hello, I'm trying to install it with 
FORCE_SUBPROJECT_SRSRAN=OFF and system-wide srsran (/usr/include/srsran and /usr/lib64/libsrs*)
```find_package(SRSRAN)``` files to fine cmake file, it is probably called FindSRSLTE.cmake,
However, there are multiple variables which are not set and it is clearly broken.

Could you fix all scripts please?

hdtuanss, 2024-04-25:
Hi, if you wanna use your own srsRAN library, the easiest way is after executing cmake command, replace the original srsRAN folder in /build/srsRAN-src/ with your own srs lib and build the project again.
Thanks!

blshkv, 2024-04-28:
Ideally, I want to build it as a share library, similar to c-mnalib.
The ```FORCE_SUBPROJECT_CMNALIB``` option works fine, the ```FORCE_SUBPROJECT_SRSRAN``` is broken.
Thus, this bug report.