SysSec-KAIST/LTESniffer #68: Compile Errors

ttfake, 2024-04-15:
Hi, 

I am receiving the following two errors during compilation:
------------------------------------------------
[  2%] Building C object cmnalib-build/src/cmnalib/CMakeFiles/cmnalib_obj.dir/src/devices/at_sierra_wireless_em7565.c.o
In file included from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/bundled/fmt/format.h:44,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/bundled/fmt/ostream.h:13,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/bundled/fmt/printf.h:14,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/detail/log_backend.h:25,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/log_channel.h:25,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/logger.h:25,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/srslog.h:26,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/support/srsran_assert.h:26,
                 from /home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/src/support/emergency_handlers.cc:23:
/home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/include/srsran/srslog/bundled/fmt/core.h:1307:44: error: field ‘value’ has incomplete type ‘std::array<char, 256>’
 1307 |     std::array<char, max_pool_string_size> value;
-----------------------------------
[  2%] Building C object cmnalib-build/src/cmnalib/CMakeFiles/cmnalib_obj.dir/src/util/gps_transform.c.o
/home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/src/asn1/liblte_mme.cc: In function ‘LIBLTE_ERROR_ENUM liblte_mme_unpack_network_name_ie(uint8**, LIBLTE_MME_NETWORK_NAME_STRUCT*)’:
/home/ouroboros/tools/LTESniffer/build/srsRAN-src/lib/src/asn1/liblte_mme.cc:2093:20: error: ‘tmp_char’ may be used uninitialized [-Werror=maybe-uninitialized]
 2093 |       if (tmp_char == 0x0A || tmp_char == 0x0D || (tmp_char >= 0x20 && tmp_char <= 0x3F) ||
------------------------------------

These errors are occurring on both Gentoo (Linux Ouroboros 6.6.21-gentoo) and Kali Linux (Linux kali 6.6.9-amd64). Gentoo GCC gcc version 13.2.1 20240210; Kali GCC gcc version 13.2.0.

For the second error, I tried unsuccessfully to disable the "Treating all warnings as errors" feature.

Any help that you can give me would be greatly appreciated.
Thanks,
Tim

blshkv, 2024-04-16:
@ttfake  I'm creating an ebuild for Pentoo (Gentoo), and we already have srsram ebuild, where the error is fixed, however, there is another issue, see https://github.com/SysSec-KAIST/LTESniffer/issues/67.

The problem seems with Hardened GCC.
As a quick workaround, remove ```-Werror``` from CMakefile.txt

blshkv, 2024-04-16:
For the first error, add ```#include <array``` to bundled/fmt/core.h

ttfake, 2024-04-16:
Great!. Thank you very much, that solved my first set of issues and got me on the road to solving two more issues. In case they help others, the string library is missing from srsRAN-src/lib/include/srsran/radio/channel_mapping.h and the cstdint library is missing from srsRAN-src/srsue/hdr/stack/rrc/nr/rrc_nr_config.h. I now I have compiled LTESniffer.

I understand how you discovered the solution for problem two and I was on the right track in trying to disable the "Treat all warnings..." but, for my own edification (and because I firmly believe in teaching a person to fish) how did you discover the resolution to problem one?  The error message seems ambiguous to me. I guess I could have thought along the lines that the compiler needs to have the full definition of the array type and it clearly wasn't being given but, until you pointed it out, I didn't pick up on that. I guess, maybe, it just comes with experience.

blshkv, 2024-04-16:
Just google: ```error: field ‘value’ has incomplete type ‘std::array<char, 256>’```

Beliwars, 2025-06-19:
Hello guys, I put it in here so anybody who encounters this issue would know what to do, my problem was I haven´t installed the Soapy library so the cmake didin´t found any of the SDR libraries and it activated the compilation flag for no RF; anyway I installed the SOAPY library and recompiled in order to not to have the compilation flag activated.

blshkv, 2025-06-20:
SoapySDR is optional, according to line:
```https://github.com/SysSec-KAIST/LTESniffer/blob/main/CMakeLists.txt#L30```