SysSec-KAIST/LTESniffer #73: Error predecoding

roy81f, 2024-05-04:
Everything was built well  ( B210 SDR ),   but Unfortunately, I am experiencing some issues, I pressed CtrlC after a second because it shows errors in a loop. Please advise me.
( Cellular Z screenshot attached also )
[image]

**sudo ./src/LTESniffer -A 2 -W 4 -f 2630e6 -C -m 0**

LTESniffer

Sat_May__4_22.04.20_2024.
Creating Phy
Creating 4 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD  zmq 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.6.0.0-0ubuntu1~jammy1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning receiver to 2630.000 MHz
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 376 FDD, CP: Normal  , DetectRatio=100% PSR=5.98, Power=-1.4 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 376 (N_id_2=1)
Setting sampling rate 23.04 MHz
Finding PSS... Peak: 1.17605, FrameCnt: 0 State: 0
Finding PSS... Peak: 4.15256, FrameCnt: 0 State: 1
Finding PSS... Peak: 1.07717, FrameCnt: 17 State: 0
Finding PSS... Peak: 1.10889, FrameCnt: 17 State: 0
Finding PSS... Peak: 1.16413, FrameCnt: 17 State: 0
Finding PSS... Peak: 1.02132, FrameCnt: 17 State: 0
Finding PSS... Peak: 1.09379, FrameCnt: 17 State: 0
Finding PSS... Peak: 3.36544, FrameCnt: 0 State: 1
Finding PSS... Peak: 1.00353, FrameCnt: 27 State: 0
Finding PSS... Peak: 1.28281, FrameCnt: 27 State: 0
Finding PSS... Peak: 1.00443, FrameCnt: 27 State: 0
Finding PSS... Peak: 1.00284, FrameCnt: 27 State: 0
Finding PSS... Peak: 5.96871, FrameCnt: 0 State: 1
Finding PSS... Peak: 1.10079, FrameCnt: 25 State: 0
Finding PSS... Peak: 1.05865, FrameCnt: 25 State: 0
Finding PSS... Peak: 1.49186, FrameCnt: 25 State: 0
Finding PSS... Peak: 1.03743, FrameCnt: 25 State: 0
Finding PSS... Peak: 1.06135, FrameCnt: 25 State: 0
Finding PSS... Peak: 7.31725, FrameCnt: 0 State: 1
 - Type:            FDD
 - PCI:             376
 - Nof ports:       4
 - CP:              Normal  
 - PRB:             100
 - PHICH Length:    Normal
 - PHICH Resources: 1/6
 - SFN:             396
Decoded MIB. SFN: 396, offset: 3
/home/roy/LTESniffer/build/srsRAN-src/lib/src/phy/mimo/precoding.c:1853: Error predecoding multiplex: not implemented for 4 Tx ports
/home/roy/LTESniffer/build/srsRAN-src/lib/src/phy/phch/pdsch.c:880: Error predecoding
/home/roy/LTESniffer/src/src/DL_Sniffer_PDSCH.cc:1104: ERROR: Decoding PDSCH
[... 281 lines cut ...]
20   35272    Unknown     2        2        0        0(0%)        0        0        0        2        0        
21   35490    Unknown     10       10       0        0(0%)        0        0        0        8        0        
22   38075    Unknown     17       17       0        0(0%)        0        0        0        15       0        
23   38367    Unknown     1        1        0        0(0%)        0        0        0        1        0        
24   39030    Unknown     11       11       0        0(0%)        0        0        0        10       0        
25   43424    Unknown     54       54       0        0(0%)        0        0        0        54       0        
26   52557    Unknown     14       14       0        0(0%)        0        0        0        14       0        
27   54488    Unknown     5        5        0        1(20%)       0        0        0        4        0        
28   54489    Unknown     10       10       0        5(50%)       0        0        0        4        0        
29   56199    Unknown     7        7        0        0(0%)        0        0        0        5        0        
30   56828    Unknown     51       51       0        0(0%)        0        0        0        47       0        
31   57120    Unknown     3        3        0        0(0%)        0        0        0        2        0        
32   57708    Unknown     11       11       0        0(0%)        0        0        0        9        0        
33   57729    Unknown     2        2        0        0(0%)        0        0        0        2        0        
34   57731    Unknown     1        1        0        0(0%)        0        0        0        1        0        
35   64326    Unknown     27       27       0        0(0%)        0        0        0        25       0        
[256Tracking] Total: 0 RNTIs are 64QAM table, 0 RNTIs are 256QAM table, 35 RNTIs are Unknown 

Destroyed Phy
Skipped subframe: 0 / 1016
nof_decoded_locations, nof_cce, nof_missed_cce, nof_subframes, nof_subframe_collisions_dw, nof_subframe_collisions_up, time, nof_locations
33342, 31589, 1403, 616, 2, 0, 0.000000, 58513
Skipped subframes: 0 (0%)
Saving MAC PCAP file
Deleted DL Sniffer core

hdtuanss, 2024-05-09:
Hi,
you are sniffing on a 4-port base station, which is not supported by LTESniffer.
Please find a 2-port base station and try again.
Thanks

closed 2024-10-23 by hdtuanss

AnilSarode, 2025-05-02:
@hdtuanss 

Appreciate your response.  

Because of the 4-port, the error is there, but we are still getting RNTI values. Do these errors affect the IMSI API to get the IMSI values?