SysSec-KAIST/LTESniffer #74: Is LTESniffer related to band?

stuitee, 2024-06-03:
When I have set EARFCN=3350, DL=2680e6，PCI=1.
LTESniffer fails to synchronize and resolve the broadcast channel.

hmily@hmily:~/LTESniffer-main/build/src$ sudo ./LTESniffer -A 2 -W 8 -f 2680e6 -C -m 0 

LTESniffer

Mon_Jun__3_13.05.46_2024.
Creating Phy
Creating 8 Worker threads 
Opening RF device with 2 RX antennas...
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.6.0.0-0-unknown
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/local/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
RF device 'UHD' successfully opened
Starting AGC thread...
Tunning receiver to 2680.000 MHz
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 277 FDD, CP: Normal  , DetectRatio=50% PSR=2.21, Power=14.1 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 277 (N_id_2=1)
Cell not found after 0 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=62% PSR=2.01, Power=13.3 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 1 (N_id_2=1)
Cell not found after 1 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 277 FDD, CP: Normal  , DetectRatio=50% PSR=2.01, Power=14.3 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 277 (N_id_2=1)
Cell not found after 2 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id: 277 FDD, CP: Normal  , DetectRatio=50% PSR=3.17, Power=14.7 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 277 (N_id_2=1)
Cell not found after 3 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=44% PSR=2.29, Power=14.0 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 1 (N_id_2=1)
Cell not found after 4 trials. Trying again (Press Ctrl+C to exit)
Searching for cell...
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
*Found Cell_id:   1 FDD, CP: Normal  , DetectRatio=83% PSR=2.23, Power=14.1 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
Decoding PBCH for cell 1 (N_id_2=1)

hdtuanss, 2024-06-03:
Hi, if you are using USRP B210, please add -a "num_recv_frames=512" to your command and run it again.
Thanks

stuitee, 2024-06-04:
It still can't be solved, the problem is the same as above, the broadcast channel can't be resolved. We are an analog base station based on srsRAN, can you test it?

closed 2024-06-04 by stuitee