SysSec-KAIST/LTESniffer #78: Cannot sniff uplink traffic

ad1tya-ahuja, 2024-07-15:
On using the exact values from "Cellular Z" app with uplink mode we are unable to capture any cell tower. Even if cell tower is found then LteSniffer cannot sniff from it
[image]

Same goes for downlink traffic.

[image]

What exactly is happening here?

tuomasOjalehto1, 2024-08-18:
I have the same problem. Is it just because of a bad reception?

hdtuanss, 2024-10-23:
Hi, could you run downlink sniffing with option -d and see the SNR? 
It could be because of low signal quality.
Thank you!