SysSec-KAIST/LTESniffer #89: BladeRF: Trying to figure out how to set arguments to start

Andrea99-beep, 2024-11-08:
Hello,

I am trying to use LTESniffer with a BladeRF device for downlink sniffing, but I would like to configure a few parameters specifically:

-     Sampling rate: How can I set the sampling rate for the BladeRF device? Is there a way to adjust the sample rate via the configuration or command line arguments?

-     Disabling SoapySDR: The program is attempting to use SoapySDR as a backend, but I want to use BladeRF directly without Soapy. How can I ensure that LTESniffer uses BladeRF and not SoapySDR? Are there any specific flags or settings I should configure?

For now I've tried in this way and I get this output:

sudo LTESniffer -a rf.device=bladeRF -A 2 -W 8 -f 806e6 -I 286 -m 0  

LTESniffer Start!!!

Creating Phy
Creating 8 Worker threads 

LTESniffer_Core: Starting...

Fri_Nov__8_17.53.39_2024.
Sniffer Mode: DL only
Available RF device list: UHD  soapy  bladeRF  zmq 
Trying to open RF device 'UHD'
Opening USRP channels=2, args: rf.device=bladeRF
Unable to open RF device 'UHD'
Trying to open RF device 'soapy'
Soapy has found device #0: backend=libusb, device=0x02:0x08, driver=bladerf, instance=0, label=BladeRF #0 [5b00f071..3d9837cb], serial=5b00f07115fa4830b0f4716f3d9837cb, 
Selecting Soapy device: 0
Setting up Rx stream with 2 channel(s)
Setting up Tx stream with 2 channel(s)
Available device sensors: 
 - RFIC_TEMP
Available sensors for Rx channel 0: 
 - PRE_RSSI
 - SYM_RSSI
Available sensors for Rx channel 1: 
 - PRE_RSSI
 - SYM_RSSI
State of gain elements for Rx channel 0 (AGC supported):
 - full: 71.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - dsa: -10.00 dB
Rx antenna set to RX
Tx antenna set to TX
RF device 'soapy' successfully opened
Opening RF device with 2 RX antennas...
Starting AGC thread...
Tunning DL receiver to 806.000
Setting sampling rate 11.52 MHz
Found PSS... NID2: 1, Peak: 1.14, FrameCnt: 0 State: 0
Missed PSS Attempts: 0
Found PSS... NID2: 1, Peak: 1.03, FrameCnt: 0 State: 0
Missed PSS Attempts: 1
Found PSS... NID2: 1, Peak: 1.13, FrameCnt: 0 State: 0

hdtuanss, 2024-11-12:
Hi, unfortunately, I also dont have any experience with bladeRF, I only have USRP to test. 
However, many users reported that Bladerf works well. So I will keep the issue open here for any advice from other users.
Thanks.

Andrea99-beep, 2024-11-12:
I'm testing using this command: LTESniffer -A 1 -W 4 -f xxxe6 -m 0 -g 30 
He prints me "Processed 1000/1000 subframes" but It doesn't find any RNTI and also the ltesniffer_dl_mode.pcap is empty, as he hasn't sniffed anything, but I'm sure there is traffic going on.

hdtuanss, 2024-11-12:
> I'm testing using this command: LTESniffer -A 1 -W 4 -f xxxe6 -m 0 -g 30 He prints me "Processed 1000/1000 subframes" but It doesn't find any RNTI and also the ltesniffer_dl_mode.pcap is empty, as he hasn't sniffed anything, but I'm sure there is traffic going on.

Could you please copy and paste your terminal log including the command was used here?