SysSec-KAIST/LTESniffer #95: External 10MHz ref signal is not working for USRP B210?

petertester2015, 2025-04-11:
When starting LTEsniffer in the following way I would expect some printout that the external ref signal is used on the SDR:

$ UHD_LOG_LEVEL=trace ./LTESniffer -f 2645000000 -m 0 -W 8 -C -g 0 -a "type=b200,clock_source=external,master_clock_rate=23.04e6" -v 2>&1 | tee log.txt

Grepping the log file show no indication of external ref signal:

$ egrep -i "(found cell)|(clock)|(ref)|(external)" log.txt
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Opening USRP channels=1, args: type=b200,clock_source=external,master_clock_rate=23.04e6
*Found Cell_id: 486 FDD, CP: Normal  , DetectRatio=100% PSR=6.87, Power=-4.8 dBm
 Found Cell_id:   0 FDD, CP: Normal  , DetectRatio= 0% PSR=0.00, Power=-inf dBm
 Found Cell_id: 374 FDD, CP: Normal  , DetectRatio=62% PSR=3.85, Power=-8.7 dBm

Doing a simple test in Python shows that there needs to be some time for the reference signal to lock, but it does work:
import uhd
import time
usrp = uhd.usrp.MultiUSRP("type=b200")
usrp.set_clock_source("external")
time.sleep(1)
ref_locked = usrp.get_mboard_sensor("ref_locked")
print("Reference locked:", ref_locked.to_bool())
print("Clock source:", usrp.get_clock_source(0))
print("Master clock rate:", usrp.get_master_clock_rate() / 1e6, "MHz")

Output:
$ python3 check_ref_lock_v2.py 
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.8.0.0-0ubuntu1~jammy1
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Reference locked: True
Clock source: external
Master clock rate: 16.0 MHz

So, using LTEsniffer with parameters to force the use of external ref signal is probably not working. If I disconnect the ref signal, the LTEsniffer works exactly the same.

(Background: I don't expect the external ref signal to be a huge difference. But before we proceed with field testing and setting up things in our measurement bus, we would like to know what benefit the 10MHz signal would give us and if we need to arrange with that in the bus as well.)

alphafox02, 2025-04-19:
I know this is from the dual usrp branch, but maybe there’s some clues here as to how it ultimately may end up using the Gpsdo for even the main branch.

I’m actually looking to buy a gpsdo(s) for the LibreSDR in an attempt to run two of them together.

https://github.com/SysSec-KAIST/LTESniffer/tree/LTESniffer-multi-usrp?tab=readme-ov-file#make-your-sdr-ready

shufu-z, 2025-04-22:
I have the same issue and it doesn't seem to work when I use an external clock divider to provide a stable GPSDO clock source.